nickg/nvc #220: Ambiguous character enumeration

rhinton, 2015-08-05:
I'm pretty sure this is an error because M*Sim doesn't complain.  I get the error "ambiguous use of enumeration literal 'Z'" when compiling the following file.

###### fred3.vhd
```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

package fred3 is
  function my_favorite_letter return character;
end package fred3;

package body fred3 is

  function my_favorite_letter return character is
  begin
    for ch in 'A' to 'Z' loop
      if ch = 'R' then
        return ch;
      end if;
    end loop;
  end function my_favorite_letter;

end package body fred3;
```

The key is that 'Z' is a member of the `std_logic` type, so I can see that it might be ambiguous.  And I can't quote chapter and verse from the LRM on why it should work.  But still, M*Sim handles it....

nickg, 2015-08-05:
I think the relevant snippet in the 2008 LRM is in section 5.3.2.2:

> Otherwise, the type of the range shall be determined by applying the rules of 12.5 to
the range, considered as a complete context, using the rules that the type shall be discrete and that both bounds shall have the same type. These rules apply also to a discrete range used in a loop parameter specification (see 10.10) or a generate parameter specification (see 11.8).

It's not ambiguous given that we know the left and right indices must be the same type and `'A'` is unambiguously `CHARACTER`.

nickg, 2015-08-05:
Should be OK in the latest git version

closed 2015-08-05 by nickg