nickg/nvc #223: some declarations are excluded from generate's block declarative part

mitchsm, 2015-09-04:
In parse.c, p_generate_statement() excludes some declarations from being allowed in the block declarative part:

```c
   if (scan(tSIGNAL, tTYPE, tSUBTYPE, tFILE, tCONSTANT, tFUNCTION, tIMPURE,
            tPURE, tALIAS, tATTRIBUTE, tBEGIN)) {
      while (not_at_token(tBEGIN))
         p_block_declarative_item(g);
      consume(tBEGIN);
   }
```

Missing from this list (but allowable in a block declarative part) are tPROCEDURE, tFOR, tCOMPONENT, tUSE, and tSHARED. As a test, I added tPROCEDURE to the list. "make check" still passes, and the following VHDL could then be compiled and executed successfully:

```VHDL
entity test is
end entity test;

use std.textio.all;

architecture test of test is
begin

    g1: if true generate
        procedure doit is
        begin
            write(OUTPUT, "OK." & LF);
        end procedure doit;
    begin
        process
        begin
            doit;
            wait;
        end process;
    end generate g1;

end architecture test;
```
Likewise, by adding tCOMPONENT to the list, I was then able to successfully declare and use a component:

```VHDL
entity sub_ent is
    port (val: out natural);
end entity sub_ent;

architecture test of sub_ent is
begin
    val <= 2;
end architecture test;

entity test is
end entity test;

use std.textio.all;

architecture test of test is
    signal val : natural;
begin

    g1: if true generate
        component sub_ent is
            port (val: out natural);
        end component sub_ent;
    begin
        c1: component sub_ent port map (val => val);
    end generate g1;

    process
    begin
        wait for 1 ns;
        write(OUTPUT, "Value is " & integer'image(val) & LF);
        wait;
    end process;

end architecture test;
```

Are those exceptions still needed? The other places where p_block_declarative_item() is called are not wrapped with similar exclusions.

nickg, 2015-09-11:
Yes you're right it's missing several tokens.

That slightly ugly check is required because the `begin` is optional in a generate statement. At that point you don't know whether to expect a `block_declarative_item` or a `concurrent_statement` without looking at the token that follows. In places where the `begin` is mandatory you don't have that problem as you just keep parsing `block_declarative_item`s until you hit a `begin`.

closed 2015-09-11 by nickg

mitchsm, 2015-09-11:
Ah, thanks for the explanation.