nickg/nvc #224: missing a couple of semantic checks for signal parameter to procedure

mitchsm, 2015-09-04:
The following (illegal) VHDL doesn't get caught by the analyser and results in an assertion in elaboration:

```VHDL
entity test is
end entity test;

architecture test of test is
    -- next line should fail to compile because it's not legal to assign a default value to a signal parameter
    procedure proc(signal a : integer := 1) is
    begin
    end procedure proc;
begin
    -- next line should also fail to compile because it's not legal to have no actual or an OPEN actual for a signal parameter
    proc(open);
end architecture test;
```

Output:
```
$ nvc -a test.vhd -e test
nvc: ../src/opt.c:132: opt_tag_last_value_fcall: Assertion `(kind == T_ARRAY_REF) || (kind == T_ARRAY_SLICE)' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib64/libc.so.6(gsignal+0x37) [0x7f24fac0d5d7]
/lib64/libc.so.6(gsignal+0x37) [0x7f24fac0d5d7]
/lib64/libc.so.6(abort+0x148) [0x7f24fac0ecc8]
/lib64/libc.so.6(+0x2e546) [0x7f24fac06546]
/lib64/libc.so.6(+0x2e5f2) [0x7f24fac065f2]
nvc() [0x44fcef]
nvc(object_visit+0x152) [0x458792]
nvc(object_visit+0x11f) [0x45875f]
nvc(object_visit+0x11f) [0x45875f]
nvc(tree_visit+0x48) [0x42e0e8]
nvc() [0x412d1d]
nvc() [0x41340b]
nvc(main+0x463) [0x411d93]
/lib64/libc.so.6(__libc_start_main+0xf5) [0x7f24fabf9af5]
nvc() [0x411e49]
-----------------------------
```

mitchsm, 2015-09-05:
Here's a fix for the first part:

```diff
diff --git a/src/sem.c b/src/sem.c
index 583bd05..7ea8000 100644
--- a/src/sem.c
+++ b/src/sem.c
@@ -1960,6 +1960,9 @@ static bool sem_check_port_decl(tree_t t)
       if (!sem_check_constrained(value, type))
          return false;
 
+      if (tree_class(t) == C_SIGNAL)
+         sem_error(t, "parameter of class SIGNAL can not have a default value");
+
       if (!type_eq(type, tree_type(value)))
          sem_error(value, "type of default value %s does not match type "
                    "of declaration %s", sem_type_str(tree_type(value)),
```

nickg, 2015-09-11:
The above all look fine to me. Can you create a pull request for them?

mitchsm, 2015-09-11:
Sorry for the confusion. These fixes were in pull request #231, which you merged a couple of days ago. This issue is closed. (I'm new to github, so it took me a bit to learn how to properly submit changes.)

closed 2015-09-11 by mitchsm

nickg, 2015-09-11:
Oh yes I forgot I'd already done that...