nickg/nvc #227: elaboration error on OPEN actual in subprogram

mitchsm, 2015-09-05:
An elaboration error occurs when a subprogram parameter is explicitly OPEN rather than simply being omitted:
```VHDL
use std.textio.all;

entity test is
end entity test;

architecture test of test is
    procedure proc(a,b : integer := 1) is
    begin
        write(OUTPUT, integer'image(a*b) & LF);
    end procedure proc;

    function calc(a,b : integer := 1) return integer is
    begin
        return a+b;
    end function calc;
begin
    process
    begin
        write(OUTPUT, integer'image(calc(a => 2)) & LF);
        write(OUTPUT, integer'image(calc(a => 2, b => open)) & LF);
    end process;

    proc(a => open, b => 2);
    -- proc(b => 2);

end architecture test;
```
results in:
```
$ nvc -a test.vhd -e test
** Fatal: cannot lower expression kind T_OPEN
	File open_const_procedure_input.vhd, Line 19
            write(OUTPUT, integer'image(calc(a => 2, b => open)) & LF);
                                                          ^^^^
```

mitchsm, 2015-09-18:
@nickg Hi Nick, I have a fix for this issue, but I'm not sure about stacking pull requests or if you'll be satisfied with these changes. Please let me know how you'd like to proceed.

nickg, 2015-09-19:
Just merged this one. I guess individual pull requests for each issue makes things less confusing, but I'm not too bothered. Thanks!

closed 2015-09-19 by nickg