nickg/nvc #234: second instance of entity causes elaboration assertion

mitchsm, 2015-09-13:
Instantiating an entity twice causes an assertion during elaboration:
```
$ nvc -a test.vhd -e test
nvc: ../src/lower.c:4505: lower_process: Assertion `!tree_has_code(proc)' failed.
[stack trace omitted]
```
Here's the testcase:
```VHDL
entity subent is
end entity subent;
architecture subentarch of subent is
    signal a : boolean;  -- commenting this out eliminates the assertion
begin
    proc: process
    begin
        report "ok";
        wait;
    end process;
end architecture;
entity test is
end entity test;
architecture testarch of test is
begin
ent1: entity work.subent;
ent2: entity work.subent;
end architecture;
```
As best as I can tell, what's happening in the failing case (with the signal declaration present) is that the copy of the entity/architecture made during elaboration is only partial -- it does not create a copy of the process object. So, the process object is shared by the elaborated :test:ent1 instance and the original WORK.SUBENT-SUBENTARCH in the library. When the second instance is elaborated, it does the same thing so it ends up with the same process object. This can be seen when dumping the vcode. Take note of the name of the process -- it's been prepended with the paths for each instance.
Alternatively, if the signal declaration is commented out, no copy is made, so when the architecture's ident is changed when elaborating the first instance, the second instance gets a completely fresh copy out of the library (it thinks it is not loaded) and there is no issue.
```
$ nvc -a 2inst_bad.vhd -e test --dump-vcode

Name       WORK.TEST.elab
Kind       context
Blocks     1
Registers  1
Types      4
Variables  2
  resolved_:test:ent1:a                 // @<0..1> => 0..1 
  resolved_:test:ent2:a                 // @<0..1> => 0..1 
Signals    2
  :test:ent1:a                          // $<0..1> => 0..1 
    Shadow resolved_:test:ent1:a
    Nets   [0]
  :test:ent2:a                          // $<0..1> => 0..1 
    Shadow resolved_:test:ent2:a
    Nets   [1]
Begin
   0: r0 := const 0                     // 0..1 => 0 
      :test:ent1:a := set initial r0
      resolved_:test:ent1:a := resolved address :test:ent1:a
      :test:ent2:a := set initial r0
      resolved_:test:ent2:a := resolved address :test:ent2:a
      return 

Name       :test:ent2::test:ent1:proc
Kind       process
Context    WORK.TEST.elab
Blocks     3
Registers  6
Types      9
Variables  0
Begin
   0: return 
   1: r0 := const 0                     // 0..3 => 0 
      // Dead heap save definition of r1 
      r2 := const 111                   // 0..183 => 111 
      r3 := const 107                   // 0..183 => 107 
      r4 := const [r2,r3]               // @<0..183> => 0..183 
      r5 := const 2                     // # => 2 
      report r4 length r5 severity r0
      wait 2
   2: jump 1

Name       :test:ent2::test:ent1:proc
Kind       process
Context    WORK.TEST.elab
Blocks     3
Registers  6
Types      9
Variables  0
Begin
   0: return 
   1: r0 := const 0                     // 0..3 => 0 
      // Dead heap save definition of r1 
      r2 := const 111                   // 0..183 => 111 
      r3 := const 107                   // 0..183 => 107 
      r4 := const [r2,r3]               // @<0..183> => 0..183 
      r5 := const 2                     // # => 2 
      report r4 length r5 severity r0
      wait 2
   2: jump 1

nvc: ../src/lower.c:4505: lower_process: Assertion `!tree_has_code(proc)' failed.
```

nickg, 2015-09-13:
You could also work around it making a useless reference to `a` in the process: `assert a = a;` or similar.

The elaboration phase only copies objects when it knows it needs to make changes to each instance: usually this is just signal and generic declarations. And then a shallow copy is made of anything that references them. But processes weren't being marked for copying explicitly so that only happened when one of the statements inside referenced a signal or generic.

closed 2015-09-13 by nickg

mitchsm, 2015-09-13:
Ah, I hadn't thought of that. Thanks for the comments on elaboration.