nickg/nvc #250: T_RECORD_REF not yet supported for offset calculation

luismarques, 2015-10-07:
When I have something like this:

```
    (...)
    type field_cell_signals is array(integer range <>, integer range <>) of inter_cell;
    signal cellio : field_cell_signals(0 to rows-1, 0 to cols-1);
    (...)
    cellio(col, row).alive_in(0) <= cellio(col-1, row-1).alive_out when (col > 0 and row > 0) else '0';
```

I get an error while elaborating:

```
$ nvc -e field_tb
** Fatal: tree kind T_RECORD_REF not yet supported for offset calculation
	File field.vhdl, Line 56
```

(Line 56 is the last line of the above excerpt)

Is there a workaround? Or a timeline to implement this?

diogratia, 2016-04-29:
Here's a _complete_ test case:
```vhdl
entity field is
end entity;

architecture fum of field is

    constant cols:  natural := 4;
    constant rows:  natural := 4;
    
    type inter_cell is record
        alive_in:   bit_vector (0 to 1);
        alive_out:  bit;
    end record;

    type field_cell_signals is array 
                (integer range <>, integer range <>) of inter_cell;
                
    signal cellio : field_cell_signals(0 to rows - 1, 0 to cols - 1);

    signal row:     integer range 0 to rows - 1 := 1;
    signal col:     integer range 0 to rows - 1 := 1;

begin
    cellio(col, row).alive_in(0) <= cellio(col - 1, row - 1).alive_out 
                        when (col > 0 and row > 0) else '0';
end architecture;
```

```
nvc -e field  
** Fatal: tree kind T_RECORD_REF not yet supported for offset calculation  
        File field.vhdl, Line 23
```
It's known to analyze, elaborate and run (demonstrating no range constraint errors) in another VHDL tool (ghdl-0.33 llvm).

nickg, 2016-04-30:
I put in a workaround which will allow it to elaborate but will run more slowly than it otherwise might

commit 3dff9ef7 (2016-05-01): More efficient grouping for record and multi-dimensional array signals. Fixes #250
closed 2016-05-01 by nickg