nickg/nvc #262: **Fatal: undefined synbol in .so, when generate native code shared library and execute unit.

ikwzm, 2015-11-08:
pkg.vhd
````pkg.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
package PKG is
    procedure SCAN(
        variable  TEXT_LINE     : inout LINE;
                  TEXT_END      : in    integer;
                  START_POS     : in    integer;
                  FOUND         : out   boolean;
                  FOUND_LEN     : out   integer
    );
end package;
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
package body  PKG is
    procedure SCAN(
        variable  TEXT_LINE     : inout LINE;
                  TEXT_END      : in    integer;
                  START_POS     : in    integer;
                  FOUND         : out   boolean;
                  FOUND_LEN     : out   integer
    ) is
        variable  len           :       integer;
        variable  char          :       character;
    begin
        len      := 1;
        for pos in START_POS+1 to text_end loop
            char := text_line(pos);
            case char is
                when NUL|SOH|STX|ETX|EOT|ENQ|ACK|BEL|
                     BS |HT |LF |VT |FF |CR |SO |SI |
                     DLE|DC1|DC2|DC3|DC4|NAK|SYN|ETB|
                     CAN|EM |SUB|ESC|FSP|GSP|RSP|USP|DEL =>
                    exit;
                when '['|']'|'{'|'}'|',' =>
                    for prev_pos in pos-1 downto START_POS loop
                        exit when (text_line(prev_pos) /= ' ');
                        len := len - 1;
                    end loop;
                    exit;
                when ':'=>
                    for prev_pos in pos-1 downto START_POS loop
                        exit when (text_line(prev_pos) /= ' ');
                        len := len - 1;
                    end loop;
                    exit;
                when '#' =>
                    for prev_pos in pos-1 downto START_POS loop
                        exit when (text_line(prev_pos) /= ' ');
                        len := len - 1;
                    end loop;
                    exit;
                when others => null;
            end case;
            len := len + 1;
        end loop;
        FOUND     := TRUE;
        FOUND_LEN := len;
    end procedure;
end package body;
````

test.vhd
````test.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
library WORK;
use     WORK.PKG.all;
entity  TEST is
end     TEST;
architecture MODEL of TEST is
begin
    process
        variable  text_line  : LINE;
        variable  text_end   : integer;
        variable  found      : boolean;
        variable  found_len  : integer;
    begin
        write(text_line, string'("{A:1}"));
        text_end  := 4;
        SCAN(text_line, text_end, 0, found, found_len);
        assert (FALSE) report "done." severity FAILURE;
        wait;
    end process;
end MODEL;
````

````
shell% nvc --work=WORK -a pkg.vhd
shell% nvc -L ./ --work=WORK -a test.vhd
shell% nvc -L ./ -e --native TEST
/usr/lib/llvm-3.6/bin/llc -relocation-model=pic /xxx/work/_WORK.TEST.final.bc -filetype=obj
/usr/bin/gcc -shared -o /xxx/work/_WORK.TEST.final.so /xxx/work/_WORK.TEST.final.o
shell% nvc -L ./ -r TEST
** Fatal: /xxx/work/_WORK.TEST.final.so:
          /xxx/work/_WORK.TEST.final.so:
          undefined symbol: line_37.PREV_POS
````

ikwzm, 2016-01-05:
An error did not occur when I revised a sentence as follows.

pkg.vhd
````VHDL:pkg.vhd
   :
             -- when '['|']'|'{'|'}'|',' => -- NG
                when ',' =>               -- OK
   :
````

p_case_statement_alternative() at parse.c  or lower_case_scalar() at lower.c has a problem?

nickg, 2016-01-10:
Yes, it was a bug with multiple alternatives in a single `when`.

commit 0e3d217c (2016-01-10): Do not generate duplicate code for case statement alternatives. Fixes #262
closed 2016-01-10 by nickg