nickg/nvc #263: library MODELSIM_LIB not found

AxelAtS, 2015-11-13:
I have modelsim_lib in an existing vhdl-testbench (for signal spy). When analysing with nvc, nvc says:

** Error: library MODELSIM_LIB not found in:
            /cygdrive/D/e/altera/14.0/modelsim_ase/modelsim_lib/util
            /cygdrive/D/e/altera/14.0/modelsim_ase/modelsim_lib
            /cygdrive/h/.nvc/lib
            /usr/local/share/nvc

My NVC-command:
nvc -L /cygdrive/D/e/altera/14.0/modelsim_ase/modelsim_lib -L "/cygdrive/D/e/altera/14.0/modelsim_ase/modelsim_lib/util\ -a /cygdrive/D/<...>/tb.vhd

Using: nvc 0.1 (llvm 3.5.2; tcl 8.5)
on: Windows 7 pro + cygwin

How can I get this not to throw an error (I do not need modelsim's signal spy functionality in this case).

Thank you for NVC!

Axel

nickg, 2015-11-15:
Is `modelsim_lib` the library directory itself? In that case you need to pass the directory containing it to `-L` as this location will be used to search for the library name e.g. `-L /cygdrive/D/e/altera/14.0/modelsim_ase/`.

AxelAtS, 2015-11-23:
OK, didn't pass the directory containing it, my mistake. - Unfortunately, this wasn't all, I think, because the library is compiled using Modelsim format. -  I do not need the functionality of of the lib, I only need that the reference to the lib doesn't crash the nvc -a . I got this working by creating a "dummy-lib" (without functionality):

* Took existing vhdl-sources from: D:\e\altera\14.0\modelsim_ase\vhdl_src\modelsim_lib and put it into folder /Modelsim_lib_modified_sources
* Removed the line "alias mti_default is default [return forcetype];...", because it crashed nvc -a
* Compiled the new source with nvc --work=modelsim_lib -a /cygdrive/D/cygwin64/home/nvc_at_home/modelsim_lib_nvc/modelsim_lib_modified_sources/mti_util.vhd
* Placed this folder into another folder, arbitrarily named modelsim_lib_nvc
* I reference modelsim_lib now this way: nvc -L /cygdrive/D/cygwin64/home/nvc_at_home/modelsim_lib_nvc

(Maybe the above is useful for somebody else.)

Thank you for your help!
Axel

closed 2015-11-24 by nickg