nickg/nvc #265: gccjit

hiyuh, 2015-11-20:
long time no see!

i expect NVC can use gccjit instead of llvm by reconstructing `src/rt` and related ones.
you're also interested in?

nickg, 2015-11-22:
I think is easier now that I've rewritten the way the code generator works since last year. You would need to add a different implementation of `src/cgen.c` and `src/rt/jit.c` and the rest should be independent (maybe). But I don't know much about gccjit so no idea how hard it would be.

hiyuh, 2015-11-23:
i don't know gccjit so much too. :)

you said adding different impl. of `src/cgen.c` and `src/rt/jit.c` for gccjit.
so you mean you preferred to impl. llvm/gccjit separately in different source files,
and switch them in build time by using  some `#if` via `configure` for the first step?

nickg, 2015-11-24:
I think they should go in separate source files e.g. under `src/gcc/*`. Then they can be compiled in separately with a configure option like `--with-backend=gcc`.

hiyuh, 2015-11-25:
introducing `--with-backend=[llvm|gcc]` looks reasonable for me too.

i think i need to carefully investigate current llvm backend independency
to plan to reconstruct `src/{llvm,gcc}/*` as you mentioned.
following simple find-grep lists some sources, any comment?
```
$ find . -name '*.c' -or -name '*.h' | xargs grep -i -e 'llvm' | sed -e 's|:.*||' | uniq
./config.h
./src/cgen.c
./src/common.c
./src/common.h
./src/elab.c
./src/link.c
./src/lower.c
./src/nvc.c
./src/phase.h
./src/rt/jit.c
./test/test_lower.c
```

nickg, 2015-11-26:
There's also `src/link.c` I forgot about which has lots of LLVM-specific code. The rest of the LLVM dependencies should be trivial - e.g. `#ifdef` on whether LLVM mangles symbol names.

hiyuh, 2015-11-26:
@nickg 
no functional changes. :)

i'm now reconsider about manage llmv-dep sources w/o `#ifdef HAVE_LLVM`,
(`src/link.c`, `src/cgen.c` and `src/rt/jit.c`).
renaming them w/ appropriate backend suffix (`*-llvm.c`) isn't better?

nickg, 2015-11-26:
I'd rather the backends go in separate `src/llvm/` and `src/gcc/` directories. I think that makes the makefiles, etc. neater. And if you have GCC disabled you can ignore the contents of the `gcc/` directory.

hiyuh, 2015-12-01:
update title for more ambitious.

hiyuh, 2015-12-02:
@nickg 
no functional change, again. :)
if it looks ok, please pull.
then i'll do bits (maybe, creating `src/stub` or `src/empty` at first?) if i had some spare time.

nickg, 2015-12-03:
I don't think we want to make these changes in `master` until another backend is somewhat functional. I've created a branch `multi-backend` - can you create a pull request to merge your changes into that?

nickg, 2018-02-06:
Is this still an interesting thing to do now we have LLVM working on Windows again?

hiyuh, 2018-02-07:
supporting alternative backend is just for fun.
b/c gccjit/libfirm support for Windows looks bad for now, IMHO.
if you mean that branch looks being too orphan/old,
feel free to purge it and close this issue too.
next time i'd like to do again, i have no problem to restart from your current master.

nickg, 2018-02-13:
OK I'll close this one for now. Because there's been a lot of changes to the backend recently, you'd probably need to start again. But now only `cgen.c` has a dependency on LLVM, so it should be easier.

closed 2018-02-13 by nickg