nickg/nvc #267: -fPIC should not be hardcoded

hiyuh, 2015-11-24:
SSIA.
to resolve this issue, adding some `-fPIC` related auto* magic, or introduce `libtool`?
```
$ ./configure --enable-fst-pthread --enable-fst-judy --enable-native --enable-vhpi --enable-werror
...
$ make
...
$ make check
...
  CC       test/vhpi/lib_vhpi1_so-vhpi1.o
test/vhpi/vhpi1.c:1:0: error: -fPIC ignored for target (all code is position independent) [-Werror]
 #include "vhpi_user.h"
 ^
...
```

hiyuh, 2015-11-24:
just for onetime check, simply removing `-fPIC` from `test/vhpi/Makmodule.am`
and rerun `make check` results error.
maybe, wrong `.so` handling in `automake`?
```
$ git diff
diff --git a/test/vhpi/Makemodule.am b/test/vhpi/Makemodule.am
index c5fec31..04b4bfb 100644
--- a/test/vhpi/Makemodule.am
+++ b/test/vhpi/Makemodule.am
@@ -3,11 +3,11 @@ if ENABLE_VHPI
 check_PROGRAMS += lib/vhpi1.so lib/vhpi2.so

 lib_vhpi1_so_SOURCES = test/vhpi/vhpi1.c
-lib_vhpi1_so_CFLAGS  = -fPIC -I$(top_srcdir)/src/vhpi $(AM_CFLAGS)
+lib_vhpi1_so_CFLAGS  = -I$(top_srcdir)/src/vhpi $(AM_CFLAGS)
 lib_vhpi1_so_LDFLAGS = -shared $(VHPI_LDFLAGS) $(AM_LDFLAGS)

 lib_vhpi2_so_SOURCES = test/vhpi/vhpi2.c
-lib_vhpi2_so_CFLAGS  = -fPIC -I$(top_srcdir)/src/vhpi $(AM_CFLAGS)
+lib_vhpi2_so_CFLAGS  = -I$(top_srcdir)/src/vhpi $(AM_CFLAGS)
 lib_vhpi2_so_LDFLAGS = -shared $(VHPI_LDFLAGS) $(AM_LDFLAGS)

 endif
```
```
$ make check
  CCLD     lib/vhpi1.so.exe
test/vhpi/lib_vhpi1_so-vhpi1.o: In function `check_error':
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:20: undefined reference to `vhpi_check_error'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:21: undefined reference to `vhpi_assert'
test/vhpi/lib_vhpi1_so-vhpi1.o: In function `startup':
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:186: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:193: undefined reference to `vhpi_register_cb'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:195: undefined reference to `vhpi_get'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:201: undefined reference to `vhpi_register_cb'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:204: undefined reference to `vhpi_get_str'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:204: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:206: undefined reference to `vhpi_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:209: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:211: undefined reference to `vhpi_handle_by_name'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:214: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:216: undefined reference to `vhpi_handle_by_name'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:219: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:221: undefined reference to `vhpi_handle_by_name'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:223: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:224: undefined reference to `vhpi_release_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:226: undefined reference to `vhpi_get'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:226: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:228: undefined reference to `vhpi_release_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:195: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:208: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:218: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:213: undefined reference to `vhpi_assert'
test/vhpi/lib_vhpi1_so-vhpi1.o: In function `end_of_sim':
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:169: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:174: undefined reference to `vhpi_get_value'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:176: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:177: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:179: undefined reference to `vhpi_release_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:180: undefined reference to `vhpi_release_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:181: undefined reference to `vhpi_release_handle'
test/vhpi/lib_vhpi1_so-vhpi1.o: In function `start_of_sim':
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:131: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:136: undefined reference to `vhpi_get_time'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:145: undefined reference to `vhpi_get_value'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:147: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:148: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:151: undefined reference to `vhpi_put_value'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:163: undefined reference to `vhpi_register_cb'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:140: undefined reference to `vhpi_assert'
[... 19 lines cut ...]
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:26: undefined reference to `vhpi_handle'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:29: undefined reference to `vhpi_handle_by_name'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:32: undefined reference to `vhpi_handle_by_name'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:41: undefined reference to `vhpi_get_value'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:44: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:52: undefined reference to `vhpi_get_value'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:55: undefined reference to `vhpi_printf'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:59: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:60: undefined reference to `vhpi_get_value'
[... cut ...]
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:86: undefined reference to `vhpi_control'
test/vhpi/lib_vhpi1_so-vhpi1.o: In function `test_bin_str':
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:45: undefined reference to `vhpi_assert'
/home/hiyuh/git-repos/nvc/test/vhpi/vhpi1.c:60: undefined reference to `vhpi_assert'
collect2: error: ld returned 1 exit status
Makefile:1209: recipe for target 'lib/vhpi1.so.exe' failed
make[1]: *** [lib/vhpi1.so.exe] Error 1
Makefile:2098: recipe for target 'check-am' failed
make: *** [check-am] Error 2
```

nickg, 2015-11-24:
This is on Cygwin right? I think the VHPI modules will need to link against `libnvcimp.a`

nickg, 2015-11-24:
Should work now

closed 2015-11-24 by nickg

mitchsm, 2015-11-25:
With this change, I get the following during a clean build:
```
[...]
  CXXLD    bin/nvc
/usr/bin/ld: cannot find -lLLVM-3.4.2
collect2: error: ld returned 1 exit status
make: *** [bin/nvc] Error 1
```
It changed from executing this:
g++  -g -O2 -rdynamic -L/usr/lib64/llvm  -lz -lpthread -lffi -ltinfo -ldl -lm      -o bin/nvc src/nvc.o lib/libnvc.a lib/libcgen.a lib/librt.a lib/libjit.a lib/liblxt.a lib/libfst.a lib/libfastlz.a -Wl,-rpath /usr/lib64/llvm -lLLVM-3.4.2  -lz -ltcl -ldl  -lreadline
to this:
g++  -g -O2   -o bin/nvc src/nvc.o lib/libnvc.a lib/libcgen.a lib/librt.a lib/libjit.a lib/liblxt.a lib/libfst.a lib/libfastlz.a -Wl,-rpath /usr/lib64/llvm -lLLVM-3.4.2  -lz -ltcl -ldl  -lreadline
or, in raw form:
from: $(AM_V_CXXLD)$(CXXLINK) $(bin_nvc_OBJECTS) $(bin_nvc_LDADD) $(LIBS)
to: $(AM_V_CXXLD)$(bin_nvc_LINK) $(bin_nvc_OBJECTS) $(bin_nvc_LDADD) $(LIBS)

bin_nvc_LINK references bin_nvc_LDFLAGS but that variable is commented out, whereas CXXLINK references AM_LDFLAGS which has all the stuff that's missing above. Hope this makes sense... I know nothing of automake!

hiyuh, 2015-11-25:
@mitchsm 
sorry for introducing your error.
maybe, following change (which i have no problem on cygwin) can help?
```
diff --git a/src/Makemodule.am b/src/Makemodule.am
index 3ea29ea..b37b5b1 100644
--- a/src/Makemodule.am
+++ b/src/Makemodule.am
@@ -48,9 +48,11 @@ if ENABLE_VHPI
 bin_nvc_LDADD += lib/libvhpi.a
 endif

+bin_nvc_LDFLAGS = $(AM_LDFLAGS)
+
 if IMPLIB_REQUIRED

-bin_nvc_LDFLAGS = -Wl,--export-all-symbols -Wl,--out-implib=lib/libnvcimp.a
+bin_nvc_LDFLAGS += -Wl,--export-all-symbols -Wl,--out-implib=lib/libnvcimp.a

 pkgdata_DATA += lib/libnvcimp.a

```

@nickg 
`make check` w/ `--enable-vhpi` passed.
FYI, using '.so.exe' for DLL looks bit uncool.$ file lib/vhpi*.so.exe
```
$ file lib/vhpi*.so.exe
lib/vhpi1.so.exe: PE32 executable (DLL) (console) Intel 80386, for MS Windows
lib/vhpi2.so.exe: PE32 executable (DLL) (console) Intel 80386, for MS Windows
```

nickg, 2015-11-25:
Oops, my bad! I only tested on Cygwin and not Linux. Applied @hiyuh's fix above.

The `so.exe` thing does look a bit weird, but it's only on test files used internally. Users can generate VHPI plugins with a `.dll` extension if they like.

mitchsm, 2015-11-25:
It's working again now, thanks!