nickg/nvc #269: nonsense error while analyzing contraint type

postoroniy, 2015-11-30:
I am getting 
** Error: right index 0 violates constraint STD.STANDARD.POSITIVE
on trivial stuff like below
```vhdl
    function func1(w1,w2:integer) return integer is
        variable xdiff   : signed(CWIDTH downto 0):=(others=>'0');
...
    begin
...
        return result;
    end func1;
```
 and no error when
```vhdl
    function func1(w1,w2:integer) return integer is
        variable xdiff   : signed(CWIDTH-0 downto 0):=(others=>'0');
...
    begin
...
        return result;
    end func1;
```

mitchsm, 2015-12-01:
here's a complete testcase:
```VHDL
package issue269 is
    constant CWIDTH : positive := 10;
    type int_vec is array (integer range <>) of integer;
    signal s1 : int_vec(CWIDTH downto 0);
end package;
```
I didn't get too far tracking it down. I think bounds_check() is OK. Instead, it seems that simplify() is changing the left dimension of the int_vec type from its original definition (of INTEGER) to that of the constant, POSITIVE. The type of the left side of the range is used in the bounds-check, which is how the right side ends up producing the false error.

nickg, 2015-12-03:
@mitchsm thanks for the test case. I think the bounds checking code was getting the type information from the wrong place though: it needs to take the index constraint from the base type `int_vec` rather than the anonymous subtype created for `s1` which will have its `left` replaced with the value of `CWIDTH` (type positive) in the simplify phase.

closed 2015-12-03 by nickg