nickg/nvc #271: support alternative shell

hiyuh, 2015-12-04:
SSIA.

TCL is one of established shells, especially for many EDA tools.
but personally, i don't like this one.
introducing `--with-shell=...` to support alternative shell is worth considering to.

also NVC always depends TCL ATM.
but `make check` or use via proper build infra doesn't need interactive shell.
introducing `--with-shell=none` for this case looks reasonably to me.

nickg, 2015-12-04:
TBH I've considered removing the TCL shell a couple of times: I don't think it's particularly useful and am not aware of anyone using it (based on zero bugs reported ;-). The TCL and readline dependency is also a pain on some platforms.

Probably the best thing to do would be to improve the VHPI implementation so that it's possible to rewrite it as a VHPI plugin rather than part of the core, and then it's trivially possible to disable at compile time. Then it would also be possible to write bindings to other languages (Python, Lua, etc.).

nickg, 2017-07-21:
This is sort of fixed, in that NVC no longer has a shell ;-)

closed 2017-07-21 by nickg