nickg/nvc #273: update tools/build-xilinx.rb

hiyuh, 2015-12-04:
Xilinx looks having no passion to support ISE anymore.
this issue proposes to update `tools/build-xilinx.rb` by
* renaming to `tools/build-xlinx-ise.rb`,
* and adding new `tools/build-xilinx-vivado.rb`.

NOTE:
due to still-available their old devices such as Spartan-3AN, Virtex-4, Virtex-5, Spartan-6 and Virtex-6 are only supported by ISE, so some users can not migrate to Vivado.
such users tend to unable to use new devices such as 7-Series, UltraScale and UltraScale+, mainly b/c of its price is simply high.
nowadays, Xilinx announces Spartan-7 and releases Vivado HLx Edition.
it sounds for no money users.

P.S.
i'm not one of above ISE stickers nor no money users.
but i usually use Vivado in these days.
so keeping ISE support looks not so promising to me.

nickg, 2015-12-04:
I agree doing the above rename is a good idea. I don't have Vivado installed at the moment so if someone else could provide the script that would be great.

FWIW we still use Spartan6 at my day job and I know of a couple of other companies that do too.

hiyuh, 2015-12-08:
as @mitchsm does renaming/adding appropriate scripts and @nickg merged them.
so it's enough to achieve the original goals to a respectable degree.
now closing...

closed 2015-12-08 by hiyuh