nickg/nvc #283: STD library srachis broken?

hiyuh, 2016-03-16:
i got,
```
$ make check
...
Running suite(s): lib
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_lib.exe
Running suite(s): ident
100%: Checks: 15, Failures: 0, Errors: 0
PASS: bin/test_ident.exe
Running suite(s): parse
100%: Checks: 35, Failures: 0, Errors: 0
PASS: bin/test_parse.exe
Running suite(s): sem
100%: Checks: 76, Failures: 0, Errors: 0
PASS: bin/test_sem.exe
Running suite(s): simplify
100%: Checks: 8, Failures: 0, Errors: 0
PASS: bin/test_simp.exe
Running suite(s): elab
100%: Checks: 26, Failures: 0, Errors: 0
PASS: bin/test_elab.exe
Running suite(s): heap
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_heap.exe
Running suite(s): hash
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_hash.exe
Running suite(s): group
100%: Checks: 11, Failures: 0, Errors: 0
PASS: bin/test_group.exe
Running suite(s): bounds
100%: Checks: 11, Failures: 0, Errors: 0
PASS: bin/test_bounds.exe
Running suite(s): value
100%: Checks: 2, Failures: 0, Errors: 0
PASS: bin/test_value.exe
Running suite(s): lower
100%: Checks: 49, Failures: 0, Errors: 0
PASS: bin/test_lower.exe
          wait1 : failed
...
```

to check why `wait1` failed, run its regression manually, and i got,
```
$ /home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1  --disable-opt -r wait1
** Error: library STD not found in:
            /home/hiyuh/.nvc/lib
            /usr/local/share/nvc

** Fatal: failed to find std library
```
maybe, NVC handles `STD`

nickg, 2016-03-16:
The `run_regr.rb` automatically adds `$(top_builddir)/lib` to the library search path. Try adding `-L/home/hiyuh/git-repos/nvc/lib` before the `-a` above. 

The regression test script should automatically print all the output from `nvc` when it fails so I'm not sure why you don't get anything.

hiyuh, 2016-03-17:
> Try adding `-L/home/hiyuh/git-repos/nvc/lib` before the `-a` above.

sorry, i forget to add appropriate `-L` as you mentioned.
```
$ /home/hiyuh/git-repos/nvc/bin/nvc  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1  --disable-opt -r wait1
Segmentation fault (core dumped)
```
i'll do more check soonish...

> ... I'm not sure why you don't get anything.

b/c it looks SIGSEGV.
NVC on cygwin defines `NO_STACK_TRACE`, so is unable to handle it ATM, IIRC.

hiyuh, 2016-03-17:
FYI,
```
$ gdb ./bin/nvc.exe
...
(gdb) r  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1  --disable-opt -r wait1
Starting program: /home/hiyuh/git-repos/nvc/bin/nvc.exe -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1  --disable-opt -r wait1
[New Thread 8392.0x2da4]
[New Thread 8392.0x3690]

Program received signal SIGSEGV, Segmentation fault.
deltaq_insert_proc (wake=0x0, delta=2738036) at src/rt/rtkern.c:1133
1133       e->wakeup_gen = wake->wakeup_gen;
(gdb) bt
#0  deltaq_insert_proc (wake=0x0, delta=2738036) at src/rt/rtkern.c:1133
#1  _sched_process (delay=2738036) at src/rt/rtkern.c:357
#2  0x00000000 in ?? ()
(gdb)
```
it looks `wake` of `deltaq_insert_proc()` = `active_proc` of `_sched_process()` is `NULL`.

nickg, 2016-03-21:
I think the `delta=` issue is a probably a bug printing 64bit numbers.

I guess this happens running any simulation? What version of llvm?

hiyuh, 2016-03-22:
> I guess this happens running any simulation?

maybe, yes.
all `bin/test_*.exe` passed, but all regressions failed w/o any particular/meaningful message except library loading failure notice w/ garbage library name.

> What version of llvm?

```
$ llvm-config.exe --version
3.7.1
$ cygcheck.exe -f /usr/bin/llvm-config.exe
libllvm-devel-3.7.1-1
```

nickg, 2016-04-04:
I tried with LLVM 3.7 on Linux and don't get this failure. Is it possible to try with another LLVM version on Cygwin that was known to work?

hiyuh, 2016-04-05:
> Is it possible to try with another LLVM version on Cygwin that was known to work?

LLVM 3.5 looks working on cygwin.
maybe, LLVM 3.7 on cygwin is simply broken?
```
$ llvm-config-3.5.exe --version
3.5.2

$ cygcheck.exe -f /usr/bin/llvm-config-3.5.exe
libllvm3.5-devel-3.5.2-2

$ ./configure --enable-fst-pthread --enable-native --enable-vhpi --with-llvm=/usr/bin/llvm-config-3.5.exe
...

$ make check
...
===================
All 13 tests passed
===================
```

nickg, 2016-04-09:
There's some code in `src/rt/jit.c` that's conditionally enabled if `LLVM_HAS_MCJIT` is defined, and this only happens when >= 3.6. It could be the MCJIT code doesn't work on Cygwin for some reason, although I can't see why. You could try elaborating and running e.g. `wait1.vhd` with `--native` and see if that works, as that will bypass JIT.

hiyuh, 2016-04-11:
as you mentioned about `LLVM_HAS_MCJIT`, i've also noticed this difference.

and here's weird result, it looks NVC fails to run if `--disable-opt` is used when elab?
```
$ cygcheck.exe -f /usr/bin/llvm-config.exe
libllvm-devel-3.7.1-1

$ /usr/bin/llvm-config.exe --version
3.7.1

$ ./configure --enable-fst-pthread --enable-native --enable-vhpi --with-llvm=/usr/bin/llvm-config.exe
....

$ make check
....
Running suite(s): lib
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_lib.exe
Running suite(s): ident
100%: Checks: 15, Failures: 0, Errors: 0
PASS: bin/test_ident.exe
Running suite(s): parse
100%: Checks: 35, Failures: 0, Errors: 0
PASS: bin/test_parse.exe
Running suite(s): sem
100%: Checks: 77, Failures: 0, Errors: 0
PASS: bin/test_sem.exe
Running suite(s): simplify
100%: Checks: 8, Failures: 0, Errors: 0
PASS: bin/test_simp.exe
Running suite(s): elab
100%: Checks: 26, Failures: 0, Errors: 0
PASS: bin/test_elab.exe
Running suite(s): heap
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_heap.exe
Running suite(s): hash
100%: Checks: 3, Failures: 0, Errors: 0
PASS: bin/test_hash.exe
Running suite(s): group
100%: Checks: 11, Failures: 0, Errors: 0
PASS: bin/test_group.exe
Running suite(s): bounds
100%: Checks: 11, Failures: 0, Errors: 0
PASS: bin/test_bounds.exe
Running suite(s): value
100%: Checks: 2, Failures: 0, Errors: 0
PASS: bin/test_value.exe
Running suite(s): lower
100%: Checks: 49, Failures: 0, Errors: 0
PASS: bin/test_lower.exe
          wait1 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1  --disable-opt -r wait1
...

$ /home/hiyuh/git-repos/nvc/bin/nvc  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1 --disable-opt -r --trace wait1
TRACE (init): _sched_process delay=2738212fs
Segmentation fault (core dumped)

$ /home/hiyuh/git-repos/nvc/bin/nvc  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1 --disable-opt

$ NVC_CYG_LIB=/home/hiyuh/git-repos/nvc/lib /home/hiyuh/git-repos/nvc/bin/nvc  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1 --native
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.elab.a -o /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.so /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp

$ NVC_CYG_LIB=/home/hiyuh/git-repos/nvc/lib /home/hiyuh/git-repos/nvc/bin/nvc  -L /home/hiyuh/git-repos/nvc/lib -a /home/hiyuh/git-repos/nvc/test/regress/wait1.vhd -e wait1 --native -r --trace wait1
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.elab.a -o /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.so /home/hiyuh/git-repos/nvc/work/_WORK.WAIT1.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp
TRACE (init): reset module WORK.WAIT1.elab
TRACE (init): reset process :wait1:line_8
TRACE (init): _sched_process delay=0ms
TRACE (init): calculate initial driver values
TRACE (init): used 0 bytes of global temporary stack
TRACE 0ms+0: begin cycle
delta   process  :wait1:line_8
TRACE 0ms+0: run process :wait1:line_8
TRACE 0ms+0: _sched_process delay=1ns
TRACE 1ns+0: begin cycle
1ns     process  :wait1:line_8
TRACE 1ns+0: run process :wait1:line_8
TRACE 1ns+0: _sched_process delay=1fs
TRACE 1000001fs+0: begin cycle
1000001fs       process  :wait1:line_8
TRACE 1000001fs+0: run process :wait1:line_8
```

nickg, 2016-04-24:
I think the `--disable-opt` behaviour is OK here: now NVC uses the LLVM API to do everything except native code generation, so you don't see it calling `opt`, `llc`, etc. when optimisations are disabled.

I'll try this on a Windows PC later in the week, see if I can figure out what's going on...

nickg, 2016-05-01:
I also get the same crash on 64-bit Windows 10, but I didn't make much progress debugging it. Are you using 64-bit Windows too? It's possibly related to this: https://llvm.org/bugs/show_bug.cgi?id=24978

hiyuh, 2016-05-02:
> Are you using 64-bit Windows too?

yes, i'm on Windows 10 Pro 64bit w/ Cygwin x86.

hiyuh, 2016-06-08:
FYI,
```
diff --git a/test/run_regr.rb b/test/run_regr.rb
index 7c51ae7..72d00ab 100755
--- a/test/run_regr.rb
+++ b/test/run_regr.rb
@@ -66,7 +66,8 @@ end
 def analyse_elab_run(t)
   cmd = "#{nvc} #{std t} -a #{TestDir}/regress/#{t[:name]}.vhd"
   cmd += " -e #{t[:name]} #{native}"
-  cmd += ' --disable-opt' unless t[:flags].member? 'opt'
+  #cmd += ' --disable-opt' unless t[:flags].member? 'opt'
+  cmd += ' --native'
   cmd += ' --cover' if t[:flags].member? 'cover'
   t[:flags].each do |f|
     cmd += " -#{f}" if f =~ /^g.*=.*$/
@@ -136,7 +137,7 @@ mkdir_p 'logs'
 Dir.chdir 'logs'

 ENV['NVC_LIBPATH'] = LibPath
-ENV['NVC_CYG_LIB'] = "#{BuildDir}/src"
+ENV['NVC_CYG_LIB'] = "#{BuildDir}/lib"

 HaveVHPI = !!ENV['HAVE_VHPI']

```
w/ above dirty change, most of regressions look ok even on Windows 10 Pro 64bit w/ Cygwin x86.
but unfortunately, it looks some attributes/records are still broken.
and maybe `vhpi3.vhd` is simply missing which is for issue #64?
```
$ make check
...
        access1 : failed (no match)
0ms+1: Report Note: 10
...
          attr5 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/attr5.vhd -e attr5  --native -r attr5
** Note: 0ms+0: Report Note: 0
        Process :attr5:line_12
        File /home/hiyuh/git-repos/nvc/test/regress/attr5.vhd, Line 16
            report integer'image(p.all'length);
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Fatal: 0ms+0: Assertion Error: Assertion violation.
        Process :attr5:line_12
        File /home/hiyuh/git-repos/nvc/test/regress/attr5.vhd, Line 17
            assert p.all'length = 10;
            ^^^^^^^^^^^^^^^^^^^^^^^^^
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/logs/attr5/work/_WORK.ATTR5.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/logs/attr5/work/_WORK.ATTR5.elab.a -o /home/hiyuh/git-repos/nvc/logs/attr5/work/_WORK.ATTR5.final.so /home/hiyuh/git-repos/nvc/logs/attr5/work/_WORK.ATTR5.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp
...
        access3 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/access3.vhd -e access3  --native -r access3
** Fatal: 0ms+0: Assertion Error: Assertion violation.
        Process :access3:line_11
        File /home/hiyuh/git-repos/nvc/test/regress/access3.vhd, Line 21
            assert s'length = 0;
            ^^^^^^^^^^^^^^^^^^^^
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/logs/access3/work/_WORK.ACCESS3.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/logs/access3/work/_WORK.ACCESS3.elab.a -o /home/hiyuh/git-repos/nvc/logs/access3/work/_WORK.ACCESS3.final.so /home/hiyuh/git-repos/nvc/logs/access3/work/_WORK.ACCESS3.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp

        access4 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/access4.vhd -e access4  --native -r access4
** Fatal: 0ms+0: Assertion Error: Assertion violation.
        Process :access4:line_11
        File /home/hiyuh/git-repos/nvc/test/regress/access4.vhd, Line 20
            assert p'length = 10;
            ^^^^^^^^^^^^^^^^^^^^^
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/logs/access4/work/_WORK.ACCESS4.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/logs/access4/work/_WORK.ACCESS4.elab.a -o /home/hiyuh/git-repos/nvc/logs/access4/work/_WORK.ACCESS4.final.so /home/hiyuh/git-repos/nvc/logs/access4/work/_WORK.ACCESS4.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp
...
        issue79 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/issue79.vhd -e issue79  --native -r issue79
** Fatal: null access dereference
        File /home/hiyuh/git-repos/nvc/test/regress/issue79.vhd, Line 38
                assert p(i).all = 2;
                       ^^^^^^^^
/usr/bin/llc -relocation-model=pic /home/hiyuh/git-repos/nvc/logs/issue79/work/_WORK.ISSUE79.final.bc -filetype=obj
/usr/bin/gcc -shared -Wl,--export-all-symbols -Wl,--out-implib=/home/hiyuh/git-repos/nvc/logs/issue79/work/_WORK.ISSUE79.elab.a -o /home/hiyuh/git-repos/nvc/logs/issue79/work/_WORK.ISSUE79.final.so /home/hiyuh/git-repos/nvc/logs/issue79/work/_WORK.ISSUE79.final.obj -L/home/hiyuh/git-repos/nvc/lib -lnvcimp
...
          vhpi3 : failed
/home/hiyuh/git-repos/nvc/bin/nvc  -a /home/hiyuh/git-repos/nvc/test/regress/vhpi3.vhd -e vhpi3  --native -r --load=/home/hiyuh/git-repos/nvc/lib/vhpi3.so.exe vhpi3
** Fatal: opening /home/hiyuh/git-repos/nvc/test/regress/vhpi3.vhd: No such file or directory

FAIL: test/run_regr.rb
...
```

hiyuh, 2016-08-01:
FYI,
```
diff --git a/test/run_regr.c b/test/run_regr.c
index 4c38ee1..291a94e 100644
--- a/test/run_regr.c
+++ b/test/run_regr.c
@@ -367,8 +367,7 @@ static bool run_test(test_t *test)
    push_arg(&args, "-e");
    push_arg(&args, "%s", test->name);

-   if (!(test->flags & F_OPT))
-      push_arg(&args, "--disable-opt");
+   push_arg(&args, "--native");

    if (test->flags & F_COVER)
       push_arg(&args, "--cover");
```
using `libllvm-devel-3.8.1-1` w/ above dirty change, most of regressions look ok even on Windows 10 Pro 64bit w/ Cygwin x86, as i reported before about 3.7.1.

faild regressions for 3.8.1 are same ones for 3.7.1; `access1`, `attr5`, `access3`, `access4` and `issue79`.

nickg, 2017-05-05:
Can you try the the current `mingw32` branch on Cygwin? This uses the new LLVM ORC JIT engine if LLVM >= 3.9. On Msys2 it can run some simple examples at least.

hiyuh, 2017-05-08:
I tried `mingw32` branch on Windows 10 Pro 64bit w/ Cygwin x86,
```
$ uname -a
CYGWIN_NT-10.0-WOW lynx 2.8.0(0.309/5/3) 2017-04-01 20:42 i686 Cygwin

$ cygcheck.exe -c libllvm-devel
Cygwin Package Information
Package              Version        Status
libllvm-devel        3.9.1-1        OK

$ git reset --hard && git pull && git checkout mingw32 && patch -p1 < ../nvc-run_regr-native.patch
...

$ ./autogen.sh && ./configure --enable-fst-pthread --enable-native --enable-vhpi
...

$ grep -n -e 'ORC' ./config.h
172:/* LLVM has ORC JIT engine */
173:#define LLVM_HAS_ORC 1

$ make check
...
```
the last command `make check` reports some failed regressions are;
`access1`, `attr5`, `access4`, `textio1`, `issue79` and `issue308`.
note, applied patch on above force to use `--native` instead of `--disable-opt` as i mentioned before.

`mingw32` branch looks not so bad, but it looks having similar issue.

nickg, 2017-05-08:
What if you revert that patch though? For me with LLVM 3.9 on MSYS2 the JIT works OK (but not loading native libraries).

hiyuh, 2017-05-09:
> What if you revert that patch though?

using `--disable-opt` looks flooding "No such file or directory".
FYI, see attached log of `git reset --hard ; make check` after commands of my prev post.
[nvc-mingw32-vanilla.zip](https://github.com/nickg/nvc/files/985490/nvc-mingw32-vanilla.zip)

nickg, 2018-01-28:
Just as an update on this bug, the latest master now more-or-less works on Cygwin with modern LLVM. There are still a few test failures though.

hiyuh, 2018-01-30:
```
$  ./autogen.sh && ./configure --enable-fst-pthread --enable-native --enable-vhpi && make
...
$ make check 2>&1 | tee log.txt
...
$ grep -e ': failed' log.txt | wc -l
6
````
yes, it looks you made NVC better.
only 6 regressions failed ATM. :)
[log.txt](https://github.com/nickg/nvc/files/1676355/log.txt)

nickg, 2018-02-04:
Cygwin should be fully working again now, and it's checked on every commit on AppVeyor:
https://ci.appveyor.com/project/nickg72247/nvc

closed 2018-02-04 by nickg