nickg/nvc #286: Acess Types for Records

peteut, 2016-04-08:
Refering to a element of a record does not require `.all.`, according to 15.1.3 of [1]:

```vhdl
entity access_bug is
end entity;

architecture test of access_bug is
type rec_t is record
    foo: integer;
end record rec_t;
type rec_ptr_t is access rec_t;
type rec_arr_t is array (integer range <>) of rec_ptr_t;
type rec_arr_ptr_t is access rec_arr_t;
procedure bug_procedure is
    variable rec_ptr: rec_arr_ptr_t;
begin
    rec_ptr(0) := new rec_t'(foo => 42);
    rec_ptr(0).all.foo := 2; -- this works
    rec_ptr(0).foo := 2; -- <-- bug here
end;
begin
end architecture;
```

The [commit](https://github.com/peteut/nvc/commit/12793224899d8867624ba572fcddc437f4de25fb) adds a tests yielding this error:
```
** Error: expected record type but found REC_PTR_T
        File ../test/lower/access_bug.vhd, Line 16
        rec_ptr(0).foo := 2; -- <-- bug here
        ^^^^^^^^^^
```
[1] The Designer's Guide to VHDL, Third Edition

nickg, 2016-04-14:
This was fixed by your #287 right?

peteut, 2016-04-14:
Yes, exactly. Thanks alot for `nvc`, I like it a lot!

closed 2016-04-14 by peteut