nickg/nvc #297: undefined symbol with --codegen

kraigher, 2016-10-15:
**other_pkg.vhd:**
```vhdl
package other_pkg is
  type vector_t is array (natural range <>) of integer;
  constant constant_vector : vector_t(1 to 0) := (others => 0);
end package;
```

**pkg.vhd:**
```vhdl
use work.other_pkg.all;

package pkg is
  procedure proc;
end package;

package body pkg is
  procedure other_proc(constant value : vector_t) is
  begin
    report integer'image(value(0));
  end procedure;
  procedure proc is
  begin
    other_proc(constant_vector);
  end procedure;
end package body;
```

**ent.vhd:**
```vhdl
use work.pkg.all;

entity ent is
end entity;

architecture a of ent is
begin
  main : process
  begin
    proc;
    wait;
  end process;
end architecture;
```

**test.sh**
```
set -ex
rm -rf work
nvc --std=2008 -a
nvc --std=2008 -a other_pkg.vhd
nvc --std=2008 -a pkg.vhd --codegen pkg
nvc --std=2008 -a ent.vhd
nvc --std=2008 -e ent
nvc --std=2008 -r ent
```

**output:**
```
~/repo/nvc_bugs/bug2$ sh test.sh 
+ rm -rf work
+ nvc --std=2008 -a
+ nvc --std=2008 -a other_pkg.vhd
+ nvc --std=2008 -a pkg.vhd --codegen pkg
/usr/lib/llvm-3.4/bin/llc -relocation-model=pic /home/kraigher/repo/nvc_bugs/bug2/work/_WORK.PKG-body.bc -filetype=obj
/usr/bin/gcc -shared -o /home/kraigher/repo/nvc_bugs/bug2/work/_WORK.PKG-body.so /home/kraigher/repo/nvc_bugs/bug2/work/_WORK.PKG-body.o
+ nvc --std=2008 -a ent.vhd
+ nvc --std=2008 -e ent
+ nvc --std=2008 -r ent
** Fatal: /home/kraigher/repo/nvc_bugs/bug2/work/_WORK.PKG-body.so: /home/kraigher/repo/nvc_bugs/bug2/work/_WORK.PKG-body.so:
          undefined symbol: WORK.OTHER_PKG.CONSTANT_VECTOR
```

nickg, 2016-10-17:
This is slightly tricky because the shared library generated for `pkg` is loaded before the rest of the code, which contains `WORK.OTHER_PKG.CONSTANT_VECTOR`, is JIT-ed. For a code relocation this would be ok as we call `dlopen` with `RTLD_LAZY`, but Linux needs to resolve all data relocations at load time, so we end up with a circular dependency.

It should be possible to detect this at the time `--codegen` is called and generate a warning/error, or automatically run generate native code for the dependent packages.

kraigher, 2016-10-18:
Ok I see. Adding `--codegen other_pkg` when analyzing other_pkg.vhd would avoid the error.
A related question, is it OK to run `--codegen` twice when a package has been split into two files one containing the package header and another containing the body? For example:
```
nvc -a pkg.vhd --codegen pkg
nvc -a pkg_body.vhd --codegen pkg
```

The reason I ran into this bug/limitation is that I was only running `--codegen` for package bodies but there was a package that did not have a body so I never did `--codegen` for that package.

nickg, 2016-10-18:
For package declarations that need code to be generated (e.g. `other_pkg` above) the shared library will be written twice which although redundant but harmless.

nickg, 2017-12-30:
This is fixed now as native compilation is always used

closed 2017-12-30 by nickg