nickg/nvc #305: "Fatal: expression cannot be folded to an integer constant" when instantiate entity using record type in generic.

ikwzm, 2016-12-27:
```
-- a_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
package TEST_TYPES is
    type WIDTH_TYPE is record
        DATA : integer;
    end record;
end package;

library ieee;
use     ieee.std_logic_1164.all;
use     work.TEST_TYPES.all;
entity  TEST_SUB is
    generic (
        WIDTH  : WIDTH_TYPE
    );
    port (
        DATA_I : in  std_logic_vector(WIDTH.DATA-1 downto 0);
        DATA_O : out std_logic_vector(WIDTH.DATA-1 downto 0)
    );
end TEST_SUB;
architecture MODEL of TEST_SUB is
begin
    DATA_O <= DATA_I;
end MODEL;

library ieee;
use     ieee.std_logic_1164.all;
use     work.TEST_TYPES.all;
entity  TEST_NG is
end TEST_NG;
architecture MODEL of TEST_NG is
    constant WIDTH  : WIDTH_TYPE := (DATA => 8);
    signal   DATA_I : std_logic_vector(WIDTH.DATA-1 downto 0);
    signal   DATA_O : std_logic_vector(WIDTH.DATA-1 downto 0);
begin
    DUT: entity WORK.TEST_SUB
        generic map (WIDTH  => WIDTH)
        port map    (DATA_I => DATA_I, DATA_O => DATA_O);
    process begin
        DATA_I <= "00000000"; 
        wait for 10 ns;
        assert(DATA_O /= "00000000") report "OK." severity NOTE;
        assert(DATA_O  = "00000000") report "NG." severity ERROR;
        DATA_I <= "00000001";
        wait for 10 ns;
        assert(DATA_O /= "00000001") report "OK." severity NOTE;
        assert(DATA_O  = "00000001") report "NG." severity ERROR;
        assert FALSE report "Simulation complete." severity FAILURE;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a a_ng.vhd
shell$ ../build/bin/nvc -L ../build/lib -e test_ng
** Warning: expression prevents constant folding
	File a_ng.vhd, Line 32
** Fatal: expression cannot be folded to an integer constant
	File a_ng.vhd, Line 18
```

Thank you for your ongoing support.
I feel that it is getting closer to my ideal step by step.

nickg, 2016-12-27:
Let me know if you have any feedback or performance issues with your real-world design

closed 2016-12-27 by nickg

ikwzm, 2016-12-28:
Thanks.

Currently, I am trying to make my own AXI BFM(Bus Function Model) with nvc.

https://github.com/ikwzm/Dummy_Plug

commit b6185303 (2016-12-27): Allow record aggregates to be constant folded. Fixes #305