nickg/nvc #307: "Error: value -13 outside bounds 0 to 2147483647 for parameter FIELD" on std.textio.write procedure.

ikwzm, 2016-12-28:
### Success Case
```
-- test_ok.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_OK is
end     TEST_OK;
architecture MODEL of TEST_OK is
    constant TIME_WIDTH : integer := 13;
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
        begin
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT,  TIME_WIDTH);
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , -TIME_WIDTH);
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for 10 ns;
        p(Now, string'("Simulation Done."));
        wait;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ok.vhd
shell$ ../build/bin/nvc -L ../build/lib -e TEST_OK
shell$ ../build/bin/nvc -L ../build/lib -r TEST_OK
         0 ns Simulation Start.
        10 ns Simulation Done.
```

### NG Case 1

```
-- test_ng_1.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_NG_1 is
    generic (TIME_WIDTH : integer := 13);
end     TEST_NG_1;
architecture MODEL of TEST_NG_1 is
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
        begin
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT,  TIME_WIDTH);
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , -TIME_WIDTH);
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for 10 ns;
        p(Now, string'("Simulation Done."));
        wait;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_1.vhd
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_1
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_1.vhd, Line 17
```

### NG Case 2

```
-- test_ng_2.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_NG_2 is
end     TEST_NG_2;
architecture MODEL of TEST_NG_2 is
    constant TIME_WIDTH : integer := 13;
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
        begin
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT, ABS(TIME_WIDTH));
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , ABS(TIME_WIDTH));
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for 10 ns;
        p(Now, string'("Simulation Done."));
        assert FALSE report "Simulation complete." severity FAILURE;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_2.vhd
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_2.vhd, Line 15
                    WRITE(text_line, T, RIGHT, ABS(TIME_WIDTH));
                                               ^^^^^^^^^^^^^^^
```

### NG Case 3

```
-- test_ng_3.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_NG_3 is
    generic (TIME_WIDTH : integer := 13);
end     TEST_NG_3;
architecture MODEL of TEST_NG_3 is
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
        begin
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT, ABS(TIME_WIDTH));
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , ABS(TIME_WIDTH));
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for 10 ns;
        p(Now, string'("Simulation Done."));
        assert FALSE report "Simulation complete." severity FAILURE;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_3.vhd
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_3
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_3.vhd, Line 15
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_3.vhd, Line 17
```

### NG Case 4

```
-- test_ng_4.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_NG_4 is
    generic (TIME_WIDTH : integer := 13);
end     TEST_NG_4;
architecture MODEL of TEST_NG_4 is
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
            variable u_time_width : integer;
        begin
            u_time_width := ABS(TIME_WIDTH);
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT, u_time_width);
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , u_time_width);
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for 10 ns;
        p(Now, string'("Simulation Done."));
        assert FALSE report "Simulation complete." severity FAILURE;
    end process;
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_4.vhd
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_4
shell$ ../build/bin/nvc -L ../build/lib -r TEST_NG_4
** Fatal: value -13 outside bounds 0 to 2147483647 for parameter FIELD
	File test_ng_4.vhd, Line 17
```

nickg, 2017-01-02:
Should be fixed now!

ikwzm, 2017-01-03:
I confirmed that NG Case 1 has been fixed.
NG Case 2-4 has not fixed yet, is it?

```
shell$ git log -1
commit ee1236082c8acf5dc760a3d6c0cd86443385981c
Author: Nick Gasson <nick@nickg.me.uk>
Date:   Thu Dec 29 21:22:38 2016 +0000

    Add test case for issue #307
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_1.vhd 
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_1
shell$ ../build/bin/nvc -L ../build/lib -r TEST_NG_1
         0 ns Simulation Start.
        10 ns Simulation Done.
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_2.vhd 
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_2.vhd, Line 15
                    WRITE(text_line, T, RIGHT, ABS(TIME_WIDTH));
                                               ^^^^^^^^^^^^^^^
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_3.vhd 
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_3
** Error: value -13 out of bounds 0 to 2147483647 for parameter FIELD
	File test_ng_3.vhd, Line 15
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng_4.vhd 
shell$ ../build/bin/nvc -L ../build/lib -e TEST_NG_4
shell$ ../build/bin/nvc -L ../build/lib -r TEST_NG_4
** Fatal: value -13 outside bounds 0 to 2147483647 for parameter FIELD
	File test_ng_4.vhd, Line 17
```

nickg, 2017-01-05:
Yes sorry, not all of them are fixed

ikwzm, 2017-01-06:
Oh, don’t be sorry!

nickg, 2017-01-08:
Should be good now!

commit ff8df8ae (2017-01-08): Fix error constant folding absolute value. Fixes #307
closed 2017-01-08 by nickg