nickg/nvc #308: The display time of write procedure is wrong

ikwzm, 2017-01-03:
```
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     std.textio.all;
entity  TEST_NG is
    generic (TIME_WIDTH : integer := 13);
end     TEST_NG;
architecture MODEL of TEST_NG is
begin
    process
        variable  text_line      : LINE;
        procedure p(T:in time;M:in string) is
        begin
            if    (TIME_WIDTH > 0) then
                WRITE(text_line, T, RIGHT,  TIME_WIDTH);
            elsif (TIME_WIDTH < 0) then
                WRITE(text_line, T, LEFT , -TIME_WIDTH);
            end if;
            WRITE(text_line, string'(" ") & M);
            WRITELINE(OUTPUT, text_line);
        end procedure;
    begin
        p(Now, string'("Simulation Start."));
        wait for  100 ns;
        p(Now, string'("Simulation Phase 1."));
        wait for  100 ns;
        p(Now, string'("Simulation Phase 2."));
        wait for 1947 ns;
        p(Now, string'("Simulation Phase 3."));
        wait for    1 ns;
        p(Now, string'("Simulation Phase 4."));
        wait for    1 ns;
        p(Now, string'("Simulation Done."));
        wait;
    end process;
end MODEL;
```

```
../build/bin/nvc -L ../build/lib/ -a test_ng.vhd
shell$ ../build/bin/nvc -L ../build/lib/ -e TEST_NG
shell$ ../build/bin/nvc -L ../build/lib/ -r TEST_NG
         0 ns Simulation Start.
       100 ns Simulation Phase 1.
       200 ns Simulation Phase 2.
      2147 ns Simulation Phase 3.
-2146.96729599999981 ns Simulation Phase 4.
-2145.96729599999981 ns Simulation Done.
```

ikwzm, 2017-01-16:
How about changing lib/std/textio.vhd as follows?
```
    procedure write (l         : inout line;
                     value     : in time;
                     justified : in side := right;
                     field     : in width := 0;
                     unit      : in time := ns )
    is
            variable value_time: time := abs(value);
            variable digit_time: time := unit;
            variable str       : string (1 to 22);
            variable pos       : natural := str'left;
            variable digit     : integer;
        begin
            if value < 0 ns then
                str(pos) := '-';
                pos := pos + 1;
            end if;

            while value_time / 10 >= digit_time loop
                digit_time := digit_time * 10;
            end loop;
            
            while (pos <= str'right) loop
                digit      := value_time / digit_time;
                value_time := value_time - digit * digit_time;
                str(pos)   := character'val(digit + character'pos ('0'));
                pos        := pos + 1;
                exit when value_time = 0 fs and digit_time <= unit;
                if digit_time = unit and pos <= str'right then
                    str(pos) := '.';
                    pos := pos + 1;
                end if;
                exit when (digit_time / 10) * 10 /= digit_time;
                digit_time := digit_time / 10;
            end loop;
            write(l, str(1 to pos-1) & unit_string(unit), justified, field);
        end procedure; 
```

nickg, 2017-01-20:
Thanks I applied it. I had intended to rewrite the textio implementation, but this is good for now.

commit faa0ea15 (2017-01-20): Apply textio patch from @ikwzm. Fixes #308
closed 2017-01-20 by nickg