nickg/nvc #311: "** Error: no visible declaration for TEST.EVENT_1", when EVENT_1 was defined as an enumerated type element in a package named TEST.

ikwzm, 2017-02-03:
```
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
package TEST is
    type  EVENT_TYPE is (EVENT_1, EVENT_2);
end TEST;

library ieee;
use     ieee.std_logic_1164.all;
use     WORK.TEST;
entity  TEST_NG is
end TEST_NG;
architecture MODEL of TEST_NG is
    constant init_event : TEST.EVENT_TYPE := TEST.EVENT_1;
begin
end MODEL;
```

```
shell$ ../build/bin/nvc -L ../build/lib -a test_ng.vhd
** Error: no visible declaration for TEST.EVENT_1
	File test_ng.vhd, Line 14
        constant init_event : TEST.EVENT_TYPE := TEST.EVENT_1;
                                                 ^^^^^^^^^^^^
```

ikwzm, 2017-02-03:
ｓcope_import_decls () is as follows.

```C:sem.c
static bool scope_import_decls(tree_t unit, bool unqual_only, bool all)
{
...
      // Make unqualified and package qualified names visible
      if (!unqual_only) {
         if (!sem_declare(decl, true))
            return false;

         ident_t pqual = ident_from(dname, '.');
         if (pqual != NULL)
            scope_insert_alias(decl, pqual);
      }
...
}
```

I avoided the problem by adding code as follows.

```C:sem.c
static bool scope_import_decls(tree_t unit, bool unqual_only, bool all)
{
...
      // Make unqualified and package qualified names visible
      if (!unqual_only) {
         if (!sem_declare(decl, true))
            return false;

         ident_t pqual = ident_from(dname, '.');
         if (pqual != NULL) {
             scope_insert_alias(decl, pqual);
             ident_t unit_name = ident_until(pqual, '.');
             if (!all && unit_name != NULL) {
                 type_t type = tree_type(decl);
                 switch (type_kind(type)) {
                     case T_ENUM:
                     {
                         const int nlits = type_enum_literals(type);
                         for (int i = 0; i < nlits; i++) {
                             tree_t  literal = type_enum_literal(type, i);
                             ident_t alias   = ident_prefix(unit_name, tree_ident(literal), '.');
                             scope_insert_alias(literal, alias);
                         }
                         break;
                     }
                     case T_PHYSICAL:
                     {
                         const int nunits = type_units(type);
                         for (int i = 0; i < nunits; i++) {
                             tree_t  unit  = type_unit(type, i);
                             ident_t alias = ident_prefix(unit_name, tree_ident(unit), '.');
                             scope_insert_alias(unit, alias);
                         }
                         break;
                     }
                     default:
                         break;
                 }
             }
         }
      }
...
}
```

I am not confident,  I hope I can be of any help to you.

ikwzm, 2017-02-04:
I'm sorry. 
There seems to be something wrong with this change.
Please ignore.

ikwzm, 2017-02-04:
Please change the next line of the source code presented last time.

```
            if (!all && unit_name != NULL) {
```

```
            if (!all && (kind == T_TYPE_DECL) && (unit_name != NULL)) {
```

nickg, 2017-02-04:
I think your change is OK except the `!all` isn't required - the names should be visible if you do `use work.test.all` for example. Thanks!

commit ec1f603e (2017-02-04): Make package qualified enumeration and physical literals visible. Fixes #311
closed 2017-02-04 by nickg