nickg/nvc #314: Xilinx libraries compilation failure

kaolpr, 2017-03-14:
Trying to compile Xilinx libraries gives: `** Fatal: variable and stored value do not have same type`

```
Name       UNISIM.SYSMONE1-SYSMONE1_V.GET_TOKEN$vuSTD.TEXTIO.LINE;SI
Kind       function
Context    UNISIM.SYSMONE1-SYSMONE1_V
Blocks     1
Registers  13
Types      11
Variables  3
  INDEX                                 // -2^31..2^31-1 
  TK_INDEX                              // -2^31..2^31-1 
  OLD_BUF                               // A<[*] : 0..183> 
Parameters 3
  r0    BUF                             // @<A<[*] : 0..183>> => A<[*] : 0..183> 
  r1    TOKEN                           // [*] : 0..183 
  r2    TOKEN_LEN                       // @<-2^31..2^31-1> => -2^31..2^31-1 
Begin
   0: @ /opt/Xilinx/Vivado/2016.4/data/vhdl/src/unisims/primitive/SYSMONE1.vhd:481:7
      r3 := load indirect r0            // A<[*] : 0..183> 
      null check r3
      r4 := all r3                      // @<[*] : 0..183> => [*] : 0..183 
      r5 := load indirect r4            // [*] : 0..183 
      r6 := uarray left r5 dim 0        // # 
      r7 := cast r6                     // -2^31..2^31-1 => 1..2^31-1 
      r8 := uarray right r5 dim 0       // # 
      r9 := cast r8                     // -2^31..2^31-1 => 1..2^31-1 
      r10 := uarray dir r5 dim 0        // 0..1 
      r11 := select r10 then r9 else r7 // -2^31..2^31-1 
      // Elided bounds check for r11 
      INDEX := store r11
      @ /opt/Xilinx/Vivado/2016.4/data/vhdl/src/unisims/primitive/SYSMONE1.vhd:482:7
      r12 := const 0                    // -2^31..2^31-1 => 0 
      // Elided bounds check for r12 
      TK_INDEX := store r12
      @ /opt/Xilinx/Vivado/2016.4/data/vhdl/src/unisims/primitive/SYSMONE1.vhd:483:7
      OLD_BUF := store r0

** Fatal: variable and stored value do not have same type

-------- STACK TRACE --------
nvc(show_stacktrace+0x29) [0x417179]
nvc(fatal_trace+0xab) [0x41726b]
nvc() [0x47079b]
nvc() [0x467eae]
nvc() [0x467c84]
nvc() [0x468124]
nvc(lower_unit+0x1dc) [0x468a4c]
nvc() [0x413fbb]
nvc(main+0x489) [0x412799]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f6d09b07830]
nvc(_start+0x29) [0x412859]
-----------------------------
```
Tested with Vivado 2015.4, 2016.1, 2016.3, 2016.4.

nickg, 2017-08-27:
Believe this is fixed by some other changes earlier this year. It works for me now with 2017.2

closed 2017-08-27 by nickg

kaolpr, 2017-08-29:
Confirmed! Thanks!