nickg/nvc #315: "** Fatal: cannot handle tree kind T_RECORD_REF in rewrite_refs", when the value of the record type field is specified in the generic map.

ikwzm, 2017-03-26:
```
-- test_sync.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_SYNC is
    generic (
        BITS : integer := 1
    );
    port (
        I    : in  std_logic_vector(BITS-1 downto 0);
        O    : out std_logic_vector(BITS-1 downto 0)
    );
end TEST_SYNC;
architecture MODEL of TEST_SYNC is
begin
    O <= I;
end MODEL;
```

```
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG is
    generic (
        DATA_BITS  : integer := 32
    );
    port (
        I_DATA     : in  std_logic_vector(DATA_BITS-1 downto 0);
        O_DATA     : out std_logic_vector(DATA_BITS-1 downto 0)
    );
end     TEST_NG;
architecture RTL of TEST_NG is
    component TEST_SYNC
        generic (BITS: integer := 1);
        port(
            I: in  std_logic_vector(BITS-1 downto 0);
            O: out std_logic_vector(BITS-1 downto 0)
        );
    end component;
    type      INFO_TYPE is record
              DATA_LO   : integer;
              DATA_HI   : integer;
              BITS      : integer;
    end record;
    function  SET_INFO return INFO_TYPE is
        variable info   : INFO_TYPE;
        variable index  : integer;
    begin
        index        := 0;
        info.DATA_LO := index;
        info.DATA_HI := index + DATA_BITS - 1;
        index        := index + DATA_BITS;
        info.BITS    := index;
        return info;
    end function;
    constant  INFO    : INFO_TYPE := SET_INFO;
    signal    i_info  : std_logic_vector(INFO.BITS-1 downto 0);
    signal    o_info  : std_logic_vector(INFO.BITS-1 downto 0);
begin
    i_info(INFO.DATA_HI downto INFO.DATA_LO) <= I_DATA;
    SYNC: TEST_SYNC generic map (
            BITS => INFO.BITS
        )
        port map(
            I    => i_info,
            O    => o_info
        );
    O_DATA <= o_info(INFO.DATA_HI downto INFO.DATA_LO);
end RTL;
```

```
shell$ nvc -a test_sync.vhd
shell$ nvc -a test_ng.vhd
shell$ nvc -e TEST_NG
** Fatal: cannot handle tree kind T_RECORD_REF in rewrite_refs
	File test_ng.vhd, Line 43
```

nickg, 2017-05-03:
Should be OK now, can you test?

closed 2017-05-03 by nickg

ikwzm, 2017-05-04:
I tried to make sure, but I failed at the time of the analysis.

-> issue #320

commit 9cdafc68 (2017-05-03): Allow folding of functions returning records. Fixes #315