nickg/nvc #321: "nvc: ../src/eval.c:442: eval_value_cmp: Assertion `lhs->kind == rhs->kind' failed." when we use a constant that substitutes the result of a function as an argument of another function.

ikwzm, 2017-05-04:
```
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
entity  TEST_NG is
end     TEST_NG;
architecture MODEL of TEST_NG is

    function  MAX(A,B:integer) return integer is begin
        if (A > B) then return A;
        else            return B;
        end if;
    end function;

    function  CALC_DATA_SIZE(WIDTH:integer) return integer is
        variable value : integer;
    begin
        value := 0;
        while (2**(value) < WIDTH) loop
            value := value + 1;
        end loop;
        return value;
    end function;

    constant  T_DATA_WIDTH      : integer := 12;
    constant  M_DATA_WIDTH      : integer := 23;
    constant  BUF_DATA_BITS     : integer := MAX(T_DATA_WIDTH, M_DATA_WIDTH);
    constant  BUF_DATA_BIT_SIZE : integer := CALC_DATA_SIZE(BUF_DATA_BITS);

begin
end;
```

```
shell$ export NVC_EVAL_VERBOSE=1
shell$ nvc -a test_ng.vhd
shell$ nvc -e test_ng
** Note: evaluate thunk for MAX
	File test_ng.vhd, Line 26
** Note: 1ad4210__MAX$III (in MAX) returned 23
** Note: MAX returned 23
	File test_ng.vhd, Line 26
** Note: evaluate thunk for CALC_DATA_SIZE
	File test_ng.vhd, Line 27
nvc: ../src/eval.c:442: eval_value_cmp: Assertion `lhs->kind == rhs->kind' failed.

*** Caught signal 6 (SIGABRT) ***

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7fafae23f267]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7fafae23f267]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fafae240eca]
/lib/x86_64-linux-gnu/libc.so.6(+0x2e03d) [0x7fafae23803d]
/lib/x86_64-linux-gnu/libc.so.6(+0x2e0f2) [0x7fafae2380f2]
nvc() [0x44e3bc]
nvc() [0x4504eb]
nvc() [0x44e6bc]
nvc() [0x450741]
nvc(eval+0x132) [0x450ff2]
nvc() [0x44a009]
nvc(object_rewrite+0x2bd) [0x45956d]
nvc(object_rewrite+0x290) [0x459540]
nvc(object_rewrite+0xf0) [0x4593a0]
nvc(tree_rewrite+0x5c) [0x430bec]
-----------------------------

Hint: you can get better stack traces by installing the libdw-dev package and reconfiguring
```

closed 2017-05-04 by nickg

ikwzm, 2017-05-04:
Failed to build 'lib/ieee/IEEE.MATH_COMPLEX'

```
shell$ cd build
shell$ make clean
 cd .. && /bin/bash /home/ichiro/work/nvc/missing automake-1.15 --foreign
running CONFIG_SHELL=/bin/bash /bin/bash ../configure --no-create --no-recursion
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... -std=gnu99
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... -lfl
checking whether yytext is a pointer... yes
checking for ranlib... ranlib
checking for inline... inline
checking for C/C++ restrict keyword... __restrict
checking how to run the C preprocessor... gcc -std=gnu99 -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking execinfo.h usability... yes
checking execinfo.h presence... yes
checking for execinfo.h... yes
checking sys/ptrace.h usability... yes
checking sys/ptrace.h presence... yes
checking for sys/ptrace.h... yes
checking sys/sysctl.h usability... yes
checking sys/sysctl.h presence... yes
checking for sys/sysctl.h... yes
checking sys/prctl.h usability... yes
checking sys/prctl.h presence... yes
checking for sys/prctl.h... yes
checking for struct stat.st_mtimespec.tv_nsec... no
checking for struct stat.st_mtim.tv_nsec... yes
checking whether the linker accepts -rdynamic... yes
checking whether C compiler accepts -fPIC... yes
checking for __attribute__((returns_nonnull))... yes
[... 120 lines cut ...]

-------- STACK TRACE --------
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7fdfca7c8267]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7fdfca7c8267]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fdfca7c9eca]
/lib/x86_64-linux-gnu/libc.so.6(+0x2e03d) [0x7fdfca7c103d]
/lib/x86_64-linux-gnu/libc.so.6(+0x2e0f2) [0x7fdfca7c10f2]
./bin/nvc() [0x44e1df]
./bin/nvc(eval+0x185) [0x4510e5]
./bin/nvc() [0x44a009]
./bin/nvc(object_rewrite+0x2bd) [0x45960d]
./bin/nvc(object_rewrite+0x290) [0x4595e0]
./bin/nvc(object_rewrite+0xf0) [0x459440]
./bin/nvc(object_rewrite+0xf0) [0x459440]
./bin/nvc(object_rewrite+0xf0) [0x459440]
./bin/nvc(object_rewrite+0xf0) [0x459440]
./bin/nvc(tree_rewrite+0x5c) [0x430bec]
-----------------------------

Hint: you can get better stack traces by installing the libdw-dev package and reconfiguring
Makefile:2531: recipe for target 'lib/ieee/IEEE.MATH_COMPLEX-body' failed
make[1]: *** [lib/ieee/IEEE.MATH_COMPLEX-body] Error 2
Makefile:941: recipe for target 'all' failed
make: *** [all] Error 2
```

nickg, 2017-05-05:
This happened when building without `make clean`, fixed now. Thanks for reporting!

commit 773bf626 (2017-05-04): Fix constant folding for integer exponentiation. Fixes #321