nickg/nvc #331: The field value of the record calculated in the procedure is wrong.

ikwzm, 2017-06-27:
Thank you for responding to issue #328 and #330.

Sorry to trouble you, but another problem occurred.

```
-- test_ng.vhd
library ieee;
use     ieee.std_logic_1164.all;
entity  TEST_NG is
    generic (
        INFO_BITS  : integer :=  1;
        INFO_1_VAL : integer :=  0
    );
    port (
        I_INFO_0   : in  std_logic_vector(INFO_BITS-1 downto 0);
        I_INFO_1   : in  std_logic_vector(INFO_BITS-1 downto 0);
        O_INFO_0   : out std_logic_vector(INFO_BITS-1 downto 0);
        O_INFO_1   : out std_logic_vector(INFO_BITS-1 downto 0)
    );
end     TEST_NG;
architecture MODEL of TEST_NG is
    type      INFO_RANGE_TYPE is record
              DATA_LO           : integer;
              DATA_HI           : integer;
    end record;
    type      VEC_RANGE_TYPE is record
              DATA_LO           : integer;        
              DATA_HI           : integer;        
              INFO_0            : INFO_RANGE_TYPE;
              INFO_1            : INFO_RANGE_TYPE;
    end record;
    function  SET_VEC_RANGE return VEC_RANGE_TYPE is
        variable  d_pos         : integer;
        variable  v             : VEC_RANGE_TYPE;
        procedure SET_INFO_RANGE(INFO_RANGE: inout INFO_RANGE_TYPE; BITS: in integer) is
        begin
            INFO_RANGE.DATA_LO  := d_pos;
            INFO_RANGE.DATA_HI  := d_pos + BITS-1;
            d_pos := d_pos + BITS;
        end procedure;
    begin
        d_pos := 0;
        v.DATA_LO := d_pos;
        SET_INFO_RANGE(v.INFO_0, INFO_BITS);
        if (INFO_1_VAL /= 0) then
            SET_INFO_RANGE(v.INFO_1, INFO_BITS);
        end if;
        v.DATA_HI := d_pos - 1;
        if (INFO_1_VAL =  0) then
            SET_INFO_RANGE(v.INFO_1, INFO_BITS);
        end if;
        return v;
    end function;
    constant  VEC_RANGE      : VEC_RANGE_TYPE  := SET_VEC_RANGE;
    signal    i_data         : std_logic_vector(VEC_RANGE.DATA_HI downto VEC_RANGE.DATA_LO);
begin
    i_data(VEC_RANGE.INFO_0.DATA_HI downto VEC_RANGE.INFO_0.DATA_LO) <= I_INFO_0;
    O_INFO_0 <= i_data(VEC_RANGE.INFO_0.DATA_HI downto VEC_RANGE.INFO_0.DATA_LO);
    INFO_1: if (INFO_1_VAL /= 0) generate
      i_data(VEC_RANGE.INFO_1.DATA_HI downto VEC_RANGE.INFO_1.DATA_LO) <= I_INFO_1;
      O_INFO_1 <= i_data(VEC_RANGE.INFO_1.DATA_HI downto VEC_RANGE.INFO_1.DATA_LO);
    end generate;
end MODEL;
```

```
shell$ export NVC_EVAL_VERBOSE=1
shell$ nvc -a test_ng.vhd
shell$ nvc -e test_ng
** Note: evaluate thunk for "/="
	File test_ng.vhd, Line 40
** Note: "/=" returned 0
	File test_ng.vhd, Line 40
** Note: evaluate thunk for "="
	File test_ng.vhd, Line 44
** Note: "=" returned 1
	File test_ng.vhd, Line 44
** Note: evaluate thunk for SET_VEC_RANGE
	File test_ng.vhd, Line 49
** Note: 24d4bc0__SET_VEC_RANGE$uVEC_RANGE_TYPE; (in SET_VEC_RANGE) returned
          *{???}
** Note: SET_VEC_RANGE returned *{DATA_LO=>0, DATA_HI=>0, INFO_0=>{DATA_LO=>1,
          DATA_HI=>1}, INFO_1=>{DATA_LO=>1, DATA_HI=>1}}
	File test_ng.vhd, Line 49
** Note: evaluate thunk for "-"
	File test_ng.vhd, Line 10
** Note: "-" returned 0
	File test_ng.vhd, Line 10
** Note: evaluate thunk for "-"
	File test_ng.vhd, Line 11
** Note: "-" returned 0
	File test_ng.vhd, Line 11
** Note: evaluate thunk for "-"
	File test_ng.vhd, Line 12
** Note: "-" returned 0
	File test_ng.vhd, Line 12
** Note: evaluate thunk for "-"
	File test_ng.vhd, Line 13
** Note: "-" returned 0
	File test_ng.vhd, Line 13
** Note: evaluate thunk for "/="
	File test_ng.vhd, Line 54
** Note: "/=" returned 0
	File test_ng.vhd, Line 54
** Error: I_DATA slice left index 1 out of bounds 0 downto 0
	File test_ng.vhd, Line 52
** Error: I_DATA slice left index 1 out of bounds 0 downto 0
	File test_ng.vhd, Line 53
```

I expected SET_VEC_RANGE to return ```*{DATA_LO=>0, DATA_HI=>0, INFO_0=>{DATA_LO=>0, DATA_HI=>0}, INFO_1=>{DATA_LO=>1, DATA_HI=>1}}```, 
but in fact SET_VEC_RANGE is returning  ```*{DATA_LO=>0, DATA_HI=>0, INFO_0=>{DATA_LO=>1, DATA_HI=>1}, INFO_1=>{DATA_LO=>1, DATA_HI=>1}}```.

As a result, an unintended boundary error has occurred.

Thanks,
ikwzm

closed 2017-07-03 by nickg

ikwzm, 2017-07-05:
Thank you for fixing.

Is not the following printf() unnecessary?
src/eval.c
```
case VALUE_RECORD:
        {
           const int nfields = type_fields(type);
           printf("nfields=%d value->length=%d\n", nfields, value->length);   // !!   line 1957
           assert(nfields == value->length);
           tree = tree_new(T_AGGREGATE);
           for (int i = 0; i < nfields; i++) {
```

nickg, 2017-07-05:
Yes it was left in by mistake, I removed it.

commit a5fcc367 (2017-07-03): Fix constant folding bug with record assignment. Fixes #331