nickg/nvc #338: Short-circuit evaluation of boolean expressions

kraigher, 2017-07-29:
```vhdl
entity bug is
end entity;

architecture a of bug is
begin
  pure_function : process
    function fail return boolean is
    begin
      assert false report "Should never be reached" severity failure;
      return true;
    end;

    procedure proc(value : boolean; expected : boolean) is
    begin
      assert value = expected severity failure;
    end;

    variable count : natural := 0;
    variable value : boolean;
  begin

    while false and fail loop
      assert false report "Should never happen" severity error;
      count := count + 1;
    end loop;

    while true or fail loop
      count := count + 2;
      exit;
    end loop;

    if false and fail then
      assert false report "Should never happen" severity error;
      count := count + 4;
    end if;

    if true or fail then
      count := count + 8;
    end if;

    loop
      exit when true or fail;
      count := count + 16;
    end loop;

    loop
      exit when false and fail;
      count := count + 32;
      exit;
    end loop;

    proc(false and fail, false);
    proc(true or fail, true);

    value := true or fail;
    assert value;
    value := false and fail;
    assert not value;

    assert true or fail;
    assert not (false and fail);

    report integer'image(count);
    assert count = 2+8+32 severity failure;
    report "Finished";
    wait;
  end process;

  impure_function : process
    variable retval : boolean := false;

    impure function fail return boolean is
    begin
      assert false report "Should never be reached" severity failure;
      retval := not retval;
      return retval;
    end;

    procedure proc(value : boolean; expected : boolean) is
    begin
      assert value = expected severity failure;
    end;

    variable count : natural := 0;
    variable value : boolean;
  begin

    while false and fail loop
      assert false report "Should never happen" severity error;
      count := count + 1;
    end loop;

    while true or fail loop
      count := count + 2;
      exit;
    end loop;

    if false and fail then
      assert false report "Should never happen" severity error;
      count := count + 4;
    end if;

    if true or fail then
      count := count + 8;
    end if;

    loop
      exit when true or fail;
      count := count + 16;
    end loop;

    loop
      exit when false and fail;
      count := count + 32;
      exit;
    end loop;

    proc(false and fail, false);
    proc(true or fail, true);

    value := true or fail;
    assert value;
    value := false and fail;
    assert not value;

    assert true or fail;
    assert not (false and fail);

    report integer'image(count);
    assert count = 2+8+32 severity failure;
    report "Finished";
    wait;
  end process;

  access_out_of_range : process
    constant s : string := "123";

    procedure proc(value : boolean; expected : boolean) is
    begin
      assert value = expected severity failure;
    end;

    variable count : natural := 0;
    variable value : boolean;
    variable pos : natural;
  begin
    pos := 4;

    while false and s(pos) = '1' loop
      assert false report "Should never happen" severity error;
      count := count + 1;
    end loop;

    while true or s(pos) = '1' loop
      count := count + 2;
      exit;
    end loop;

    if false and s(pos) = '1' then
      assert false report "Should never happen" severity error;
      count := count + 4;
    end if;

    if true or s(pos) = '1' then
      count := count + 8;
    end if;

    loop
      exit when true or s(pos) = '1';
      count := count + 16;
    end loop;

    loop
      exit when false and s(pos) = '1';
      count := count + 32;
      exit;
    end loop;

    proc(false and s(pos) = '1', false);
    proc(true or s(pos) = '1', true);

    value := true or s(pos) = '1';
    assert value;
    value := false and s(pos) = '1';
    assert not value;

    assert true or s(pos) = '1';
    assert not (false and s(pos) = '1');

    report integer'image(count);
    assert count = 2+8+32 severity failure;
    report "Finished";
    wait;
  end process;

  non_static_access_out_of_range : process
    constant s : string := "123";

    procedure proc(value : boolean; expected : boolean) is
    begin
      assert value = expected severity failure;
    end;

    variable count : natural := 0;
    variable value : boolean;
    variable pos : natural;
  begin
    pos := 4;

    while pos < 4 and s(pos) = '1' loop
      assert false report "Should never happen" severity error;
      count := count + 1;
    end loop;

    while pos = 4 or s(pos) = '1' loop
      count := count + 2;
      exit;
    end loop;

    if pos < 4 and s(pos) = '1' then
      assert false report "Should never happen" severity error;
      count := count + 4;
    end if;

    if pos = 4 or s(pos) = '1' then
      count := count + 8;
    end if;

    loop
      exit when pos = 4 or s(pos) = '1';
      count := count + 16;
    end loop;

    loop
      exit when pos < 4 and s(pos) = '1';
      count := count + 32;
      exit;
    end loop;

    proc(pos < 4 and s(pos) = '1', false);
    proc(pos = 4 or s(pos) = '1', true);

    value := pos = 4 or s(pos) = '1';
    assert value;
    value := pos < 4 and s(pos) = '1';
    assert not value;

    assert pos = 4 or s(pos) = '1';
    assert not (pos < 4 and s(pos) = '1');

    report integer'image(count);
    assert count = 2+8+32 severity failure;
    report "Finished";
    wait;
  end process;

end;
```

```
nvc -a bug.vhd && nvc -e bug -r
** Fatal: array index 4 outside bounds 1 to 3
	File bug.vhd, Line 212
```

Not sure what the standard says about this both other simulators do short-circuiting and it is nice to have. Sorry for the long example but I do not know the corner cases of NVC in terms or how much common code there is to handle these cases.

kraigher, 2017-07-29:
GHDL output for the above test: 
```
ghdl -a bug.vhd && ghdl --elab-run bug
bug.vhd:64:5:@0ms:(report note): 42
bug.vhd:66:5:@0ms:(report note): Finished
bug.vhd:131:5:@0ms:(report note): 42
bug.vhd:133:5:@0ms:(report note): Finished
bug.vhd:192:5:@0ms:(report note): 42
bug.vhd:194:5:@0ms:(report note): Finished
bug.vhd:253:5:@0ms:(report note): 42
bug.vhd:255:5:@0ms:(report note): Finished
```

kraigher, 2017-07-29:
I found another example which cause a memory corruption inside NVC:
```vhdl
entity bug is
end entity;

architecture a of bug is
begin
  main : process
    procedure proc(s : string; variable value : out integer) is
    begin
      if s = "" then
        value := 0;
      else
        value := 1;
      end if;
    end;

    function func(s : string) return integer is
    begin
      if s = "" then
        return 0;
      end if;

      return 1;
    end;

    constant s0 : string := "foobar";
    constant s1 : string := "";

    variable value : integer;
  begin
    assert func(s1) = 1;
    assert func(s0) = 0;

    proc(s1, value);
    assert value = 1;

    proc(s0, value);
    assert value = 0;
    wait;
  end process;
end;
```

```
+ rm -rf work
+ nvc -a bug.vhd
+ nvc -e bug -r
nvc: ../src/eval.c:514: eval_value_cmp: Assertion `lhs->kind == rhs->kind' failed.

*** Caught signal 6 (SIGABRT) ***

[0x7fa2771c2428] (/lib/x86_64-linux-gnu/libc-2.23.so) ../sysdeps/unix/sysv/linux/raise.c:54 raise
[0x7fa2771c4029] (/lib/x86_64-linux-gnu/libc-2.23.so) /build/glibc-bfm8X4/glibc-2.23/stdlib/abort.c:89 abort
[0x7fa2771babd6] (/lib/x86_64-linux-gnu/libc-2.23.so) /build/glibc-bfm8X4/glibc-2.23/assert/assert.c:92 __assert_fail_base
[0x7fa2771bac81] (/lib/x86_64-linux-gnu/libc-2.23.so) /build/glibc-bfm8X4/glibc-2.23/assert/assert.c:101 __assert_fail
[0x450103] ../src/eval.c:514 eval_value_cmp
[0x4519d7] ../src/eval.c:1136 eval_vcode
[0x452ce8] ../src/eval.c:919 eval_op_fcall
[0x452667] ../src/eval.c:1728 eval_vcode
[0x453235] ../src/eval.c:2052 eval
[0x44aecf] ../src/simp.c:119 simp_tree
[0x45babc] ../src/object.c:492 object_rewrite
[0x45ba8f] ../src/object.c:412 object_rewrite
[0x45b8ee] ../src/object.c:418 object_rewrite
[0x45ba8f] ../src/object.c:412 object_rewrite
[0x45b8ee] ../src/object.c:418 object_rewrite
[0x45b8ee] ../src/object.c:418 object_rewrite
[0x43194b] ../src/tree.c:1168 tree_rewrite
[0x44c292] ../src/simp.c:1040 simplify
[0x449c44] ../src/elab.c:1452 elab_entity_arch
[0x44a3ac] ../src/elab.c:1582 elab
[0x412c29] ../src/nvc.c:307 process_command
[0x411aa8] ../src/nvc.c:1039 main
```

The assert is actually triggered from `eval_op_memcmp` in `eval.c`:
```c
static void eval_op_memcmp(int op, eval_state_t *state)
{
   value_t *dst = eval_get_reg(vcode_get_result(op), state);
   value_t *lhs = eval_get_reg(vcode_get_arg(op, 0), state);
   value_t *rhs = eval_get_reg(vcode_get_arg(op, 1), state);
   value_t *len = eval_get_reg(vcode_get_arg(op, 2), state);

   dst->kind    = VALUE_INTEGER;
   dst->integer = 1;

   EVAL_ASSERT_VALUE(op, lhs, VALUE_POINTER);
   EVAL_ASSERT_VALUE(op, rhs, VALUE_POINTER);

   for (int i = 0; i < len->integer; i++) {
      if (eval_value_cmp(&(lhs->pointer[i]), &(rhs->pointer[i]))) {
         dst->integer = 0;
         return;
      }
   }
}
``` 
The `len->integer` is to large for the `rhs` which cause pointer access out of bounds.

This seems to be caused by lines 812-813 in `lower.c`:
```c
         vcode_reg_t eq_reg = emit_cmp(VCODE_CMP_EQ, r0_len, r1_len);
         return emit_and(eq_reg, emit_memcmp(r0_data, r1_data, r0_len));
```
It seems it emits a comparison of the array lengths that should short circuit to inhibit the memcmp if the lengths are not equal. However the short-circuiting does not occur thus causing the crash.

kraigher, 2017-07-30:
A more complete test than the original post containing divide by zero and expensive pure function call.

```vhdl
entity bug is
end entity;

architecture a of bug is
  signal pos : natural := 4;
  signal divisor : natural := 0;
begin
  pure_function : process
    function fail return boolean is
    begin
      assert false report "Should never be reached" severity failure;
      return true;
    end;

    procedure proc(value : boolean; expected : boolean) is
    begin
      assert value = expected severity failure;
    end;

    variable count : natural := 0;
    variable value : boolean;
  begin

    while false and fail loop
      assert false report "Should never happen" severity error;
      count := count + 1;
    end loop;

    while true or fail loop
      count := count + 2;
      exit;
    end loop;

    if false and fail then
      assert false report "Should never happen" severity error;
      count := count + 4;
    end if;

    if true or fail then
      count := count + 8;
    end if;

    loop
      exit when true or fail;
      count := count + 16;
    end loop;

    loop
      exit when false and fail;
      count := count + 32;
      exit;
    end loop;

    proc(false and fail, false);
    proc(true or fail, true);

    value := true or fail;
    assert value;
    value := false and fail;
    assert not value;

    assert true or fail;
    assert not (false and fail);

    report integer'image(count);
    assert count = 2+8+32 severity failure;
    report "Finished";
[... 193 lines cut ...]
    wait;
  end process;

  pure_but_expensive : process
    function expensive return boolean is
      variable result : integer;
    begin
      for i in 0 to 2**29 loop
        for j in 0 to 2**29 loop
          for k in 0 to 2*29 loop
            result := ((result+k-i+j) mod 2**19) * (k mod 2**10);
          end loop;
        end loop;
      end loop;
      return (result mod 16) = 0;
    end;
  begin
    assert true or expensive;
    assert not (false and expensive);
    wait;
  end process;

end;

```

nickg, 2017-07-30:
It's definitely a bug, the LRM says:

> In general, operands in an expression are evaluated before being associated with operators. For certain operations, however, the right-hand operand is evaluated if and only if the left-hand operand has a certain value. These operations are called short-circuit operations. The binary logical operations and, or, nand, and nor defined for operands of types BIT and BOOLEAN are all short-circuit operations; furthermore, these are the only short-circuit operations.

Thanks for the comprehensive test case.

closed 2017-07-30 by nickg

kraigher, 2017-07-30:
The second example I provided with the string comparison still fails for the same reason I described.

kraigher, 2017-07-30:
Basically everything in `lower.c` which does emit_and or emit_or might need to be reviewed:
```c
         // Fails when lengths are not equal due to memory out of bounds access in emit_memcp due to non-short circuit
         vcode_reg_t eq_reg = emit_cmp(VCODE_CMP_EQ, r0_len, r1_len);
         return emit_and(eq_reg, emit_memcmp(r0_data, r1_data, r0_len));
```

nickg, 2017-07-30:
Yes sorry I missed the second part, it's fixed now.

commit 6cc33a70 (2017-07-30): Add short circuit evaluation of and, or, nor. Fixes #338