nickg/nvc #352: "000000"-x;

fdedinec, 2017-09-11:
With the code attached, and the following command,
nvc  -a flopoco.vhdl  -e TestBench_FixRealKCM_F400_uid2  -r --exit-severity=failure --wave=TestBench_FixRealKCM_F400_uid2.fst --stop-time=170ns
I get 
** Fatal: index -2147483648 violates constraint bounds 0 to 2147483647
	File ./lib/synopsys/std_logic_arith.vhd, Line 250
[bug.zip](https://github.com/nickg/nvc/files/1293226/bug.zip)

I am not too sure of my VHDL coding style, maybe the libs are wrong, but it worked with ghdl (I think) and in any cases the error message could be improved (and it smells like a signed versus unsigned bug).
Thanks for your attention and keep up the good work.

diogratia, 2017-09-15:
Your code does indeed simulate in ghdl.

Here's a simpler reproducer:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
-- use ieee.numeric_std.all;
-- use ieee.std_logic_unsigned.all;

entity FixRealKCM_F400_uid2 is
    port (
        X : in   std_logic_vector(3 downto 0);
        R : out  std_logic_vector(5 downto 0)
    );
end entity;

architecture arch of FixRealKCM_F400_uid2 is
    signal FixRealKCM_F400_uid2_Rtemp :  std_logic_vector(4 downto 0);
begin
   -- R <= "000000" - (FixRealKCM_F400_uid2_Rtemp(4) & FixRealKCM_F400_uid2_Rtemp(4 downto 0));
   R <= std_logic_vector(unsigned'(unsigned'("00000") - unsigned (FixRealKCM_F400_uid2_Rtemp(4) & FixRealKCM_F400_uid2_Rtemp(4 downto 0))));
end architecture;
```

Shows up using package std_logic_arith with either   

```vhdl
"-"[unsigned, unsigned return unsigned] 
```
or 
```vhdl 
"-"[unsigned, unsigned return std_logic_vector]
```
 there may be others. 

Doesn't show up with package numeric_std. Appears to be due to the CONV_UNSIGNED function call used on the parameter to make_binary.

nickg, 2017-09-17:
It's caused by the wrong calculation of the bounds of the slice and concatenation result.

commit a268bf28 (2017-09-17): Fix computation of bounds of array slice. Fixes #352
closed 2017-09-17 by nickg