nickg/nvc #353: Error during NVC compilation under Cygwin

Zileus, 2017-09-15:
Hello,

I'm trying to compile NVC under Cygwin. 

I cloned the github repository. Then checked the tools needed. Then run the following commands:
`./autogen.sh`
`./tools/fetch-ieee.sh`
`mkdir build && cd build`
`../configure`
 but for the `make `command, here is the result:

`$ make`
`  CC       src/util.o`
`../src/util.c: Dans la fonction « run_program »:`
`../src/util.c:1310:17: attention : implicit declaration of function « spawnv » [-Wimplicit-function-declaration]`
`    int status = spawnv(_P_WAIT, args[0], args);`
`                 ^~~~~~`
`../src/util.c:1310:24: erreur : « _P_WAIT » undeclared (first use in this function)`
`    int status = spawnv(_P_WAIT, args[0], args);`
`                        ^~~~~~~`
`../src/util.c:1310:24: note : each undeclared identifier is reported only once for each function it appears in`
`Au plus haut niveau:`
`../src/util.c:394:12: attention : « color_fprintf » defined but not used [-Wunused-function]`
` static int color_fprintf(FILE *f, const char *fmt, ...)`
`            ^~~~~~~~~~~~~`
`make[1]: *** [Makefile:1531: src/util.o] Error 1`
`make: *** [Makefile:925: all] Error 2`

Could please tell how to fix this?

Regards

nickg, 2017-09-20:
Can you try now? I merged some changes for mingw which should also improve cygwin

Zileus, 2017-09-20:
Hi,

I will give you a feedback asap ! 

Thanks anyway!

Zileus, 2017-09-20:
perfect it works.... but the script ./tools/fetch_ieee.sh is faulty.

I had to manually download all the vhdl files. Maybe a issue should be open for this script....

nickg, 2017-09-20:
> but the script ./tools/fetch_ieee.sh is faulty.

What error do you get? I ran this on Cygwin just now and it was ok...

Zileus, 2017-09-20:
The script returns that the comparison between the "reference" and the compute shasum is incorrect (line 35 of the script).
 I can't say if its becase the files are not reachable or something else is happening.

nickg, 2017-09-20:
Does your Cygwin have the `shasum` command?

nickg, 2017-10-19:
I think Cygwin is working OK now, just need to install shasum

closed 2017-10-19 by nickg