nickg/nvc #354: stack dump: ** Fatal: select arguments are not the same type

fmhess, 2017-10-31:
-- The following testbench causes nvc to dump during elaboration.

library IEEE;
use ieee.std_logic_1164.all;

entity nvc_bug is
end nvc_bug;
     
architecture behav of nvc_bug is
	signal byte : std_logic_vector(7 downto 0);
	signal byte_too : std_logic_vector(7 downto 0);

begin

	-- nvc doesn't like the byte_too(1) in the next line
	byte(1 downto 0) <= (1 => byte_too(1), 0 => '0') when true else (others => 'Z'); 
	process
	begin
		byte_too(0) <= '0';
		byte_too(1) <= '1';
		wait for 100ns;
		assert byte(0) = '0';
		assert byte(1) = '1';
				
		assert false report "end of test" severity note;
		wait;
	end process;
end behav;

fmhess, 2017-10-31:
and here is the output I see from running nvc:

$ nvc -a nvc_bug.vhd     
$ nvc -e nvc_bug     

Name       :nvc_bug:line_16
Kind       process
Context    WORK.NVC_BUG.elab
Blocks     5
Registers  40
Types      14
Variables  0
Begin
   0: @ nvc_bug.vhd:16:0
      r0 := nets :nvc_bug:byte          // $<[8] : 0..8> => [8] : 0..8 
      r1 := const 1                     // # => 1 
      r2 := const 1                     // -2^31..2^31-1 => 1 
      r3 := const 7                     // -2^31..2^31-1 => 7 
      r4 := const 6                     // -2^31..2^31-1 => 6 
      r5 := const 6                     // # => 6 
      r6 := add r0 + r5                 // $<[8] : 0..8> => [8] : 0..8 
      r7 := const 8                     // # => 8 
      r8 := const 2                     // # => 2 
      alloc driver nets r0+r7 driven r6+r8
   1: @ nvc_bug.vhd:16:0
      r9 := heap save                   // # 
      r10 := const 0                    // -2^63..2^63-1 => 0 
      r11 := const 1                    // -2^31..2^31-1 => 1 
      r12 := const 0                    // -2^31..2^31-1 => 0 
      r13 := const 1                    // 0..1 => 1 
      r14 := const 0                    // 0..1 => 0 
      r15 := nets :nvc_bug:byte         // $<[8] : 0..8> => [8] : 0..8 
      @ nvc_bug.vhd:16:5
      r16 := const 7                    // -2^31..2^31-1 => 7 
      r17 := const 0                    // # => 0 
      r18 := const 1                    // # => 1 
      // Elided dynamic bounds check for r11 
      @ nvc_bug.vhd:16:0
      // Elided dynamic bounds check for r12 
      r19 := const 6                    // -2^31..2^31-1 => 6 
      r20 := const 6                    // # => 6 
      r21 := add r15 + r20              // $<[8] : 0..8> => [8] : 0..8 
      // Begin dynamic aggregrate line 16 
      r22 := const 2                    // # => 2 
      r23 := alloca r22                 // @<0..8> => 0..8 
      r24 := alloca                     // @<#> => # 
      r24 := store indirect r17
      jump 2
   2: r25 := load indirect r24          // # 
      r26 := cmp r25 >= r22
      cond r26 then 4 else 3
   3: r27 := load resolved_:nvc_bug:byte_too@WORK.NVC_BUG.elab // @<0..8> => 0..8 
      r28 := const 0                    // # => 0 
      r29 := const 1                    // -2^31..2^31-1 => 1 
      r30 := const 7                    // -2^31..2^31-1 => 7 
      r31 := const 6                    // -2^31..2^31-1 => 6 
      r32 := const 6                    // # => 6 
      r33 := add r27 + r32              // @<0..8> => 0..8 
      r34 := const 2                    // 0..8 => 2 
      r35 := const 0                    // -2^31..2^31-1 => 0 
      r36 := const -1                   // -2^31..2^31-1 => -1 
      r37 := const 1                    // # => 1 
      r38 := cmp r25 == r37
      r39 := select r38 then r34 else r33 // 0..8 
   4: Empty basic block

** Fatal: select arguments are not the same type

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x56177ac9c056]
nvc(fatal_trace+0xa0) [0x56177ac9c130]
nvc(+0x77d87) [0x56177acfad87]
nvc(+0x668c2) [0x56177ace98c2]
nvc(+0x6ba55) [0x56177aceea55]
nvc(lower_unit+0x1d9) [0x56177acf1249]
nvc(+0x159fe) [0x56177ac989fe]
nvc(main+0x4a2) [0x56177ac97752]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f03f85f12b1]
nvc(_start+0x2a) [0x56177ac9781a]
-----------------------------

Hint: you can get better stack traces by installing the libdw-dev package and reconfiguring

nickg, 2017-11-01:
Should be fixed now, thanks for reporting

commit dbaeb60a (2017-11-01): Fix crash with certain array aggregates. Fixes #354
closed 2017-11-01 by nickg