nickg/nvc #356: feeding to_bitvector result to case crashes in analysis

fmhess, 2017-11-05:
-- The following testbench causes nvc to dump during analysis.

library IEEE;
use ieee.std_logic_1164.all;

entity nvc_bug is
end nvc_bug;

architecture behav of nvc_bug is
signal mode : std_logic_vector(1 downto 0);

begin

process
begin

	--nvc doesn't like the to_bitvector() below, fails in analysis.
	case to_bitvector(mode) is
		when "00" =>
		when "01" =>
		when "10" =>
		when "11" =>
		when others =>
	end case;
			
	assert false report "end of test" severity note;
	wait;
end process;
end behav;

fmhess, 2017-11-05:
$ nvc -a nvc_bug.vhd 
** Fatal: invalid type kind T_UARRAY in range_of

-------- STACK TRACE --------
nvc(show_stacktrace+0x16) [0x55e640acb286]
nvc(fatal_trace+0xa0) [0x55e640acb360]
nvc(range_of+0x60) [0x55e640b0c100]
nvc(+0x5eae8) [0x55e640b10ae8]
nvc(object_visit+0x163) [0x55e640b131b3]
nvc(object_visit+0x12f) [0x55e640b1317f]
nvc(object_visit+0x12f) [0x55e640b1317f]
nvc(tree_visit+0x48) [0x55e640ae8078]
nvc(+0x15e08) [0x55e640ac7e08]
nvc(main+0x4a2) [0x55e640ac6932]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f5466c6b2b1]
nvc(_start+0x2a) [0x55e640ac69fa]
-----------------------------

Hint: you can get better stack traces by installing the libdw-dev package and reconfiguring

diogratia, 2017-11-06:
The case expression is lacking a locally static subtype, during analysis there's no way to determine that the various choices match the subtype of the case expression. What's missing for the above code is an error message telling you so instead of a crash.

There's an associated bug revealed by taking the above code and providing a locally static subtype to the case expression:

```vhdl
library IEEE;
use ieee.std_logic_1164.all;

entity nvc_bug is
end nvc_bug;

architecture behav of nvc_bug is
    signal mode:  std_logic_vector(1 downto 0);

begin
    process
        subtype modetype is bit_vector(mode'range);
    begin
        case modetype'(to_bitvector(mode)) is
            when "00" =>
            when "01" =>
            when "10" =>
            when "11" =>
            when others =>
        end case;
        
        assert false report "end of test" severity note;
        wait;
    end process;
end behav;
```
nvc -a nvc_bug.vhdl
** Fatal: type kind T_UARRAY does not have item I_DIMS
```
-------- STACK TRACE --------
0   nvc                                 0x000000010e7eaa86 show_stacktrace + 54
1   nvc                                 0x000000010e7eaa39 fatal_trace + 153
2   nvc                                 0x000000010e8367f2 object_lookup_failed + 98
3   nvc                                 0x000000010e8083da type_dim + 154
4   nvc                                 0x000000010e834413 bounds_check_case + 1907
5   nvc                                 0x000000010e837462 object_visit + 882
6   nvc                                 0x000000010e8373a3 object_visit + 691
7   nvc                                 0x000000010e8373a3 object_visit + 691
8   nvc                                 0x000000010e806fe7 tree_visit + 71
9   nvc                                 0x000000010e7e6ae6 process_command + 1878
10  nvc                                 0x000000010e7e6203 main + 1187
11  libdyld.dylib                       0x00007fff8a8845c9 start + 1
12  ???                                 0x0000000000000003 0x0 + 3
-----------------------------
```
Where the code providing a locally static subtype to the case expression would be expected to analyze.  

You could note there's more than one way to provide a locally static subtype. Using a variable with a locally static type assigned with the converted value of mode analyzes without error:

```vhdl
library IEEE;
use ieee.std_logic_1164.all;

entity nvc_bug is
end nvc_bug;

architecture behav of nvc_bug is
    signal mode:  std_logic_vector(1 downto 0);

begin
    process
        variable bitmode:   bit_vector(mode'range);
        -- subtype modetype is bit_vector(mode'range);
    begin
        bitmode := to_bitvector(mode);
        case bitmode is
        -- case modetype'(to_bitvector(mode)) is
            when "00" =>
            when "01" =>
            when "10" =>
            when "11" =>
            when others =>
        end case;
        
        assert false report "end of test" severity note;
        wait;
    end process;
end behav;
```

nickg, 2017-11-07:
The first example now produces an error message. The second example from @diogratia still crashes at the moment.

closed 2017-11-07 by nickg