nickg/nvc #359: crash when signal and procedure have same name

fmhess, 2017-12-16:
Don't have a testbench for it yet, but can probably come up with one if needed.  Basically, I called a procedure called "host_write" which called an overload procedure of "host_write" and passed a signal argument to the overloaded procedure also called "host_write".  There were no warnings from analysis before the crash.  The crash went away when I renamed the signal "host_write_sig".

nvc -a ../../src/util/dma_fifos.vhd ../test_common.vhd dma_fifos_testbench.vhd
nvc: src/sem.c:5189: sem_check_ref: Assertion `zero_arg_fn || (tree_kind(next) == T_ENUM_LIT)' failed.

*** Caught signal 6 (SIGABRT) ***

[0x7f5d2b158fcf] (/lib/x86_64-linux-gnu/libc-2.24.so) ../sysdeps/unix/sysv/linux/raise.c:51 raise
[0x7f5d2b15a3f9] (/lib/x86_64-linux-gnu/libc-2.24.so) /build/glibc-p3Km7c/glibc-2.24/stdlib/abort.c:89 abort
[0x7f5d2b151e36] (/lib/x86_64-linux-gnu/libc-2.24.so) /build/glibc-p3Km7c/glibc-2.24/assert/assert.c:92 __assert_fail_base
[0x7f5d2b151ee1] (/lib/x86_64-linux-gnu/libc-2.24.so) /build/glibc-p3Km7c/glibc-2.24/assert/assert.c:101 __assert_fail
[0x55c181e04c1e] src/sem.c:5189 sem_check_ref
[0x55c181dfc3c7] src/sem.c:7431 sem_check
[0x55c181e00501] src/sem.c:1500 sem_check_constrained
[0x55c181dfc165] src/sem.c:3309 sem_check
[0x55c181dfe954] src/sem.c:2481 sem_check_stmts
[0x55c181dfd798] src/sem.c:2740 sem_check
[0x55c181dfe954] src/sem.c:2481 sem_check_stmts
[0x55c181dfdcee] src/sem.c:3103 sem_check
[0x55c181dd3609] src/nvc.c:155 process_command
[0x55c181dd21e1] src/nvc.c:1041 main

nickg, 2017-12-26:
I couldn't reproduce the crash above but there's a bug where no error is produced if a procedure `foo` is declared after a signal named `foo` in the same region (this is illegal in VHDL). I've fixed this in the commit 60fcef11. Do you still have the version of your project that triggered the original crash?

fmhess, 2017-12-27:
Here's a test program that produces the crash, using nvc git hash d38a3f4f3a69127d930635272fa716cd5c71df5d

<pre>
-- this test program aborts during analysis

library IEEE;
use ieee.std_logic_1164.all;

entity nvc_bug is
end nvc_bug;

architecture behav of nvc_bug is
	signal host_write : std_logic;
begin

	process
		procedure host_write is
		begin
		end host_write;
	begin
		host_write <= '1';
	end process;

end behav;
</pre>

fmhess, 2017-12-27:
I just did a pull from git, and the crash still happens.

nickg, 2017-12-27:
Thanks for the test case. It produces this error now:

```
** Error: invalid use of procedure HOST_WRITE
        File ../test/sem/issue359a.vhd, Line 15
      host_write <= '1';
      ^^^^^^^^^^
```

commit 4fb3f6ec (2017-12-27): Fix crash when procedure and signal have same name. Fixes #359
closed 2017-12-27 by nickg