nickg/nvc #365: OS X 10.11 build failure for 1.3.0

ilovezfs, 2018-01-01:
https://jenkins.brew.sh/job/Homebrew%20Core%20Pull%20Requests/15688/version=el_capitan/testReport/junit/brew-test-bot/el_capitan/install_nvc/

```
  CC       src/object.o
  CC       src/lower.o
  CC       src/vcode.o
  CC       src/rt/rtkern.o
src/rt/rtkern.c:411:22: error: use of undeclared identifier 'CLOCK_MONOTONIC'
   if (clock_gettime(CLOCK_MONOTONIC, &ts) != 0)
                     ^
1 error generated.
make[1]: *** [src/rt/rtkern.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

Builds fine on 10.12 and 10.13.

nickg, 2018-01-01:
Hi, could you test the patch above? It's on the `issue365` branch. I found this on SO but don't have an OS X machine to test.

ilovezfs, 2018-01-01:
@nickg yes that seems to work :)

ilovezfs, 2018-01-01:
@nickg and while I have your attention … https://github.com/nickg/nvc/issues/366

nickg, 2018-01-01:
On the master branch now. I can make a 1.3.1 release once we've fixed the other issue.

closed 2018-01-01 by nickg

ilovezfs, 2018-01-01:
Awesome! Thanks :)