nickg/nvc #366: nvc attempts to hard-code Homebrew clang shim path

ilovezfs, 2018-01-01:
https://jenkins.brew.sh/job/Homebrew%20Core%20Pull%20Requests/15688/version=high_sierra/console

```
iMac-TMP:Homebrew joe$ brew test -vd nvc
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/nvc.rb
Testing nvc
/usr/bin/sandbox-exec -f /tmp/homebrew20180101-36247-1i9l4z7.sb /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/bin/ruby -W0 -I /usr/local/Homebrew/Library/Homebrew -- /usr/local/Homebrew/Library/Homebrew/test.rb /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/nvc.rb -vd
/usr/local/Homebrew/Library/Homebrew/test.rb (Formulary::FromPathLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/nvc.rb
==> Cloning https://github.com/suoto/vim-hdl-examples.git
Updating /Users/joe/Library/Caches/Homebrew/nvc--vim-hdl-examples--git
git config remote.origin.url https://github.com/suoto/vim-hdl-examples.git
git config remote.origin.fetch +refs/heads/master:refs/remotes/origin/master
==> Checking out revision c112c17f098f13719784df90c277683051b61d05
git checkout -f c112c17f098f13719784df90c277683051b61d05 --
HEAD is now at c112c17... Small changes to help hdlcc CI
git reset --hard c112c17f098f13719784df90c277683051b61d05
HEAD is now at c112c17 Small changes to help hdlcc CI
==> /usr/local/Cellar/nvc/1.3.0/bin/nvc -a /tmp/nvc-test-20180101-36248-19s9db9/basic_library/very_common_pkg.vhd
/usr/local/Homebrew/Library/Homebrew/shims/super/clang -bundle -flat_namespace -undefined dynamic_lookup -o /private/tmp/nvc-test-20180101-36248-19s9db9/work/_WORK.VERY_COMMON_PKG.so /private/tmp/nvc-test-20180101-36248-19s9db9/work/_WORK.VERY_COMMON_PKG.o
/usr/local/Homebrew/Library/Homebrew/shims/super/clang:34:in `escape': no implicit conversion of nil into String (TypeError)
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:34:in `initialize'
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:365:in `new'
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:365:in `<main>'
** Fatal: /usr/local/Homebrew/Library/Homebrew/shims/super/clang failed with
          status 1
/usr/local/Homebrew/Library/Homebrew/test.rb (Formulary::FormulaLoader): loading /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/python.rb
/usr/local/Homebrew/Library/Homebrew/debrew.rb:11:in `raise'
BuildError: Failed executing: /usr/local/Cellar/nvc/1.3.0/bin/nvc -a /tmp/nvc-test-20180101-36248-19s9db9/basic_library/very_common_pkg.vhd
1. raise
2. ignore
3. backtrace
4. irb
5. shell
Choose an action: 5
When you exit this shell, you will return to the menu.
bash-3.2$ /usr/local/Cellar/nvc/1.3.0/bin/nvc -a /tmp/nvc-test-20180101-36248-19s9db9/basic_library/very_common_pkg.vhd
/usr/local/Homebrew/Library/Homebrew/shims/super/clang -bundle -flat_namespace -undefined dynamic_lookup -o /private/tmp/nvc-test-20180101-36248-19s9db9/work/_WORK.VERY_COMMON_PKG.so /private/tmp/nvc-test-20180101-36248-19s9db9/work/_WORK.VERY_COMMON_PKG.o
/usr/local/Homebrew/Library/Homebrew/shims/super/clang:34:in `escape': no implicit conversion of nil into String (TypeError)
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:34:in `initialize'
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:365:in `new'
	from /usr/local/Homebrew/Library/Homebrew/shims/super/clang:365:in `<main>'
** Fatal: /usr/local/Homebrew/Library/Homebrew/shims/super/clang failed with
          status 1
bash-3.2$
```

I can hack around this with
```
    # avoid hard-coding the superenv shim path
    # cc_bare = Utils.popen_read("sh -ilc 'which #{ENV["CC"]}'").chomp
    # inreplace "configure", 'cc_bare="$(which ${CC%% *})"', "cc_bare=\"#{cc_bare}\""
```
But that's not something we want to maintain in the Homebrew formula for ever.

Build log:
https://gist.github.com/ilovezfs/886fb8189c1a43a605fb2f98f08b28a5

Note the
```
==> ENV["CC"] is clang
==> which(ENV["CC"]) is /usr/local/Homebrew/Library/Homebrew/shims/super/clang
```

nickg, 2018-01-01:
Would it help if there was a configure option like `--with-system-cc=` that you could pass with the path to clang?

ilovezfs, 2018-01-01:
Yes it would be far preferable to doing string replacements, though we'd still have to do the
```
cc_bare = Utils.popen_read("sh -ilc 'which #{ENV["CC"]}'").chomp
```
and then
```
--with-system-cc=#{cc_bare}
```

It might not be terrible for us to hard code
```
--with-system-cc=/usr/bin/clang
```

None of that would be needed, though, if you sanitized the PATH before doing the `which $CC`
```
bash-3.2$ which $CC
/usr/local/Homebrew/Library/Homebrew/shims/super/clang
bash-3.2$ env -i PATH=/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which $CC
/usr/bin/clang
```

nickg, 2018-01-01:
I feel that passing it as an option to `configure` is a bit cleaner as it also allows cases where someone has installed clang in a non-default location. Could you try the patch above?

ilovezfs, 2018-01-01:
Yup the patch works. I passed `--with-system-cc=/usr/bin/clang`

closed 2018-01-01 by ilovezfs

nickg, 2018-01-01:
I made a 1.3.1 release you can use here: https://github.com/nickg/nvc/releases/tag/r1.3.1

ilovezfs, 2018-01-01:
Thanks! Voilà https://github.com/Homebrew/homebrew-core/pull/22323