nickg/nvc #368: procedure parameter in/out directions ignored in package declaration

fmhess, 2018-01-14:
<pre>
-- this package has mis-labeled "in" and "out" parameters in the procedure declaration
-- but no errors or warnings are generated.

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package nvc_package_bug is
	procedure in_through_the_out_door (
		signal sig_in: out std_logic; --note this is labeled an "out" parameter but is really "in"
		signal sig_out: in std_logic --note this is labeled an "in" parameter but is really "out"
	);
end nvc_package_bug;

package body nvc_package_bug is
	procedure in_through_the_out_door (
		signal sig_in: in std_logic;
		signal sig_out: out std_logic) is
	begin
		sig_out <= sig_in;
	end procedure in_through_the_out_door;
end package body nvc_package_bug;
</pre>

fmhess, 2018-01-14:
<pre>
-- this test program compiles and runs fine in spite of the mis-labled in/out parameters in
-- the procedure in from the nvc_package_bug package.

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.nvc_package_bug.all;

entity nvc_bug is
end nvc_bug;
     
architecture behav of nvc_bug is
	signal sig_a: std_logic;
	signal sig_b: std_logic;
begin

	process
	begin
		sig_a <= '1';
		sig_b <= '0';
		wait for 1 ns;
		in_through_the_out_door(sig_a, sig_b);
		wait for 1 ns;
		assert sig_a = '1';
		assert sig_b = '1';
		wait;
	end process;
end behav;
</pre>

nickg, 2018-01-27:
Now gives an error like this:
```
** Error: parameter SIG_IN of subprogram body WORK.NVC_PACKAGE_BUG.IN_THROUGH_THE_OUT_DOOR
          with mode IN does not match mode OUT in specification
        File test.vhd, Line 17
      signal sig_in: in std_logic;
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Note: parameter SIG_IN was originally declared here
        File test.vhd, Line 10
      signal sig_in: out std_logic; --note this is labeled an "out" parameter but i
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Error: parameter SIG_OUT of subprogram body WORK.NVC_PACKAGE_BUG.IN_THROUGH_THE_OUT_DOOR
          with mode OUT does not match mode IN in specification
        File test.vhd, Line 18
      signal sig_out: out std_logic) is
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Note: parameter SIG_OUT was originally declared here
        File test.vhd, Line 11
      signal sig_out: in std_logic --note this is labeled an "in" parameter but is 
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

commit f666ff42 (2018-01-27): Improve checking that subprogram body matches spec. Fixes #368
closed 2018-01-27 by nickg