nickg/nvc #372: Configurations don't work correctly / implementation incomplete.

donnie-j, 2018-02-13:
A sticking point for simulation of the full J-Core / S-Core DSP SoC systems (as opposed to the J-Core cpu itself) is the extensive use of configuration to select e.g. technology variants.

I need to make a set of test cases, but in general this functionality is just incomplete.  More to come (when I get a little time).

J.

nickg, 2018-02-18:
I've made some improvements to configuration parsing but there's still lots of missing features in this area. It would be good to get some "real world" test cases.

nickg, 2022-01-25:
The 1.6 release contains a lot of improvements in this area to the point where it can run the J-core CPU simulations which use configurations extensively. Please open a new ticket if you encounter further issues.

closed 2022-01-25 by nickg