nickg/nvc #377: Simulation Error

xXplutoXx, 2018-06-22:
Hi,

I observed a case where NVC delivers a wrong result (compared to questa). 
If you simulate the attached VHSL using 

nvc -a FPAdd_Test.vhdl -e FPAdd_Testbench -r --stop-time=100ns -w

the result will be different from the (correct) result obtained from questa (see attached screenshots).
Any idea?

[FPAdd_Test.vhdl.zip](https://github.com/nickg/nvc/files/2129194/FPAdd_Test.vhdl.zip)

[image]
[image]

nickg, 2018-07-06:
I made some progress on this. The difference is caused by the following line where the implicit `"="` operator is called rather than the one explicitly defined in the `ieee.std_logic_unsigned` package.

```vhdl
sticky <= '0' when (shiftedFracY_d1(23 downto 0)=CONV_STD_LOGIC_VECTOR(0,23)) else '1'; 
```

nickg, 2018-07-06:
This should be fixed now, I checked the value of `R` now matches Questa, but I'd be grateful if you could check the rest of the simulation runs as expected? Note you'll have to add `--relax=prefer-explicit` when analysing the source files. This is because the way `std_logic_unsigned` overloads the `"="` operator isn't strictly allowed by the LRM, but due to a bug this wasn't detected before.

closed 2018-07-06 by nickg

xXplutoXx, 2018-07-09:
Hi Nick,

thank you so much for the fix. Just came back from vacation and will check it ASAP.

And thank you for providing NVC! It made a very good job for us already!

Martin

commit 443e783d (2018-07-06): Fix hiding of implicitly declared operators. Fixes #377