nickg/nvc #378: Unable to run ./tools/fetch_ieee.sh

hazardlabs, 2018-07-12:
This program will download IEEE library sources from
  https://standards.ieee.org/downloads/1076/1076.2-1996
  https://raw.githubusercontent.com/tgingold/ghdl/master/libraries/vital2000
and modify them by uncommenting the definition of operator XNOR.

math_complex-body.vhdl
Check failed!

Am I doing anything wrong?
I've tried both V1.3.0 and V1.3.1 with the same results.
This is on ubuntu 18.04

nickg, 2018-07-13:
This was fixed in master (see #375) but the 1.3 release still has the old URL. I guess the best thing would be to release a new 1.4 version.

nickg, 2018-07-16:
There's now a 1.4 release with the updated script: https://github.com/nickg/nvc/releases/tag/r1.4.0

closed 2018-07-16 by nickg