nickg/nvc #383: Configure failing with LLVM 5.0.1

marph91, 2019-04-09:
I'm trying to install nvc and got stuck at executing `../configure`. It fails when detecting llvm. I'm using `opensuse 15` with `llvm 5.0.1`.

config.log:
```
configure:7411: checking for LLVM (engine bitreader bitwriter ipo linker)
configure:7433: g++ -o conftest -g -O2 -I/usr/include -fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=0 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection  -fno-strict-aliasing -fPIC -fvisibility-inlines-hidden -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -ffunction-sections -fdata-sections -O2 -g -DNDEBUG  -fno-exceptions -DLLVM_BUILD_GLOBAL_ISEL -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS   -L/usr/lib64  conftest.cpp -lz -lm -ldl  -lLLVM  >&5
g++: error: unrecognized command line option '-Wstring-conversion'; did you mean '-Wint-conversion'?
configure:7433: $? = 1
```

When removing this check manually, the configure and later the build succeed. How to fix it?

nickg, 2019-08-19:
Sorry I missed this one. Do you still have this issue? Which version of gcc do you have?

marph91, 2019-08-20:
I can reproduce the error, but when looking again at the error message, `g++` is used instead of `clang++`. The workaround is to set the environment variable correctly first, like `CXX=clang++ ../configure`. Maybe on other systems the variable is set differently?

closed 2019-08-20 by marph91