nickg/nvc #385: library 29IEEE not found

xXplutoXx, 2019-08-15:
Since updating to nvc 1.5 on Mac OS, I have a strange issue for designs using std_logic_arith (it appeared for larger designs an I was not able so far to isolate a small design where this happens):

During simulation, I get like this:

	Process :testbench_fpmult_8_26_8_26_8_26_uid2_f400_uid3_f400_uid69:test:line_1210
	File ../lib/synopsys/std_logic_arith.vhd, Line 255
** Error: library 29IEEE not found in:
            /Users/<my home>/.nvc/lib
            /usr/local/share/nvc
          
It looks to me like a kind or parser error as when I create a folder named "29IEEE" and put the "IEEE" content inside, everything works as expected (this is my current workaround to continue the work).
Any idea?

nickg, 2019-08-15:
Are you able to build from the master branch and see if the error still occurs? Also it would be useful if you could a stack trace when that "library 29IEEE not found in" error is reported. It comes from `lib_find` in `lib.c` (line 455 in master). If you could put a breakpoint there in gdb and send the stack trace that would be great.

nickg, 2021-04-03:
Closing this due to lack of activity. The library loading code has changed somewhat since 2019 so it might be fixed already. If not please reopen and try to get a stack trace from gdb as described above.

closed 2021-04-03 by nickg

nickg, 2021-04-09:
I think I found the cause of this. It should be fixed in 1.5.1 release.