nickg/nvc #386: *** Caught signal 11 (SIGSEGV) [address=0xa, ip=0x7fcfe220df8c] ***

fmhess, 2019-08-18:
Hi, I'm getting segfaults running some of my tests.  The errors look like this:

```
nvc -r frontend_cb7210p2_testbench --wave 
** Note: writing FST waveform data to frontend_cb7210p2_testbench.fst

*** Caught signal 11 (SIGSEGV) [address=0xa, ip=0x7fcfe220df8c] ***

[0x7fcfe220df8c] (/usr/local/nvc/share/nvc/ieee/_IEEE.NUMERIC_STD-body.so) IEEE.NUMERIC_STD.TO_01(25IEEE.NUMERIC_STD.UNSIGNEDL)25IEEE.NUMERIC_STD.UNSIGNED
```

I'm using git HEAD:
commit 3e4e690b167c3ccc1201602f04084c3e6656851a (HEAD -> master, origin/master, origin/HEAD)

nickg, 2019-08-19:
Can you try running with `nvc -r --trace`? That should show the process which is executing when the crash happens. Hopefully that can narrow down the cause a bit.

fmhess, 2019-08-19:
I'll do you one better, here is a test program so you can reproduce it yourself.  

issue386.vhd
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package interface_function_common is

function is_unrecognized_primary_command (byte : std_logic_vector) return boolean;

end interface_function_common;

package body interface_function_common is

	function is_unrecognized_primary_command (byte : std_logic_vector(7 downto 0)) return boolean is
		variable unsigned_stripped_byte : unsigned(7 downto 0);
	begin
		unsigned_stripped_byte(6 downto 0) := to_01(unsigned(byte(6 downto 0)));
		unsigned_stripped_byte(7) := '0';
		return unsigned_stripped_byte = X"00";
	end is_unrecognized_primary_command;
end package body interface_function_common;

--

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.interface_function_common.all;

entity remote_message_decoder is
	port(
		bus_DIO_inverted : in std_logic_vector(7 downto 0);
		unrecognized_primary_command : out std_logic
	);
 
end remote_message_decoder;
 
architecture remote_message_decoder_arch of remote_message_decoder is

begin

	
	unrecognized_primary_command <= '1' when 
			is_unrecognized_primary_command(not bus_DIO_inverted) else
		'0';
	
end remote_message_decoder_arch;

--

library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.interface_function_common.all;

entity remote_message_decoder_testbench is
end remote_message_decoder_testbench;
     
architecture behav of remote_message_decoder_testbench is
	signal bus_DIO_inverted : std_logic_vector(7 downto 0);
	signal unrecognized_primary_command : std_logic;
	
	begin
	my_decoder: entity work.remote_message_decoder 
		port map (
			bus_DIO_inverted => bus_DIO_inverted,
			unrecognized_primary_command => unrecognized_primary_command
		);
	
	process		
	begin
		bus_DIO_inverted <= X"ff";

		wait for 100ns;	
		assert false report "end of test" severity note;
		wait;
	end process;
end behav;
```
It segfaults when I run it with:
```
nvc -a issue386.vhd
nvc -e remote_message_decoder_testbench 
nvc -r remote_message_decoder_testbench
```
Oh and one other piece of info, this issue is a regression.  It used to work with an older version of nvc (probably a year or two ago).

nickg, 2019-08-20:
Thanks for reducing it! The problem here is a mismatch between the specification of `is_unrecognized_primary_command` whose argument is an unconstrained array type `std_logic_vector` and the body where it's a constrained array type `std_logic_vector(7 downto 0)`. We should report an error here when analysing the package body. The runtime crash happens because nvc passes these parameters differently (a pointer wrapped in a struct with array bounds vs a raw pointer).

I'm not sure why it used to work previously: I guess it was an accident of the way nvc merged all the generated LLVM bitcode into one module which was then optimised. Now we're generating one `.so` per translation unit.

fmhess, 2019-08-20:
Thanks, fixing the declaration prevents the segfault for me too.

nickg, 2019-08-22:
This now gives an analysis error:

```
** Error: type of parameter BYTE does not match type BIT_VECTOR in specification
	File ../test/sem/issue386.vhd, Line 7
        function is_unrecognized_primary_command (byte : bit_vector(7 downto 0)) re
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
** Note: parameter BYTE was originally declared here
	File ../test/sem/issue386.vhd, Line 2
        function is_unrecognized_primary_command (byte : bit_vector) return boolean
                                                  ^^^^^^^^^^^^^^^^^
```

commit f48ab406 (2019-08-22): Missing error when body does not match specification. Fixes #386
closed 2019-08-22 by nickg