nickg/nvc #391: [feature request/question] Reusability of testsuite

Nic30, 2019-09-17:
Hello,

I would like to use tests in your repository as a tests for [VHDL/SV parser platform](https://github.com/Nic30/hdlConvertor/tree/sv2017).

Problem is that I can not easy to automatically determine if the .vhd file is correct or error lines because it is in .c file and parsing of such a error messages may be unreliable or too complex.

Do you like the idea that one day the tests can use some files with meta-informations?
In Icarus they have .list files (https://github.com/steveicarus/ivtest/blob/master/regress-sv.list) I am not saying it is perfect but in some situations it is cleaner solution.

Maybe one day it could be possible to have some complacence tests for VHDL tools like there is https://github.com/SymbiFlow/sv-tests for a SystemVerilog.

nickg, 2019-09-19:
I'm not opposed to this, and we already sort of have that for tests that generate runtime errors (`test/regress/gold/`). But I wonder whether this set of tests meets your needs, as they were developed in a rather ad-hoc way as I encountered bugs during development.

You might instead want to look at VESTS: it's an old test suite for VHDL-87/93. I think originally it was funded by the US air force. It has the advantage that it's much more comprehensive and complete than the nvc tests, and it's organised into "compliant" and "non complaint" tests, which is what you're looking for here.

The official website has been down from a long time, but I've got a snapshot of the last CVS version on GitHub that I occasionally use to test nvc. It's been really good at finding parser/analyser corner cases.

https://github.com/nickg/vests

The interesting tests are under `vhdl-93/billowitch`, the `ashenden` directory is just examples from a textbook.

nickg, 2020-04-24:
Closing this now, see the answer above. If you have any other questions feel free to email me.

closed 2020-04-24 by nickg