nickg/nvc #392: Segmentation fault when running elaborated unit

joda4711, 2020-04-15:
Hello,

I have a hard time getting NVC to work at all.

Compiling from source seemingly works fine, but actually running a
unit results in nothing at all with frequent segmentation faults.

This is my test source.

```VHDL
library ieee;
use ieee.std_logic_1164.all;

entity test is
end entity;

architecture rtl of test is
  signal foo : std_logic;
begin
  foo <= '0';
end architecture;
```

Elaborate and run like this (more or less directly from the README):

```Shell
$ nvc -a test.vhdl -e test -r -w  test
/usr/bin/ld -shared -o /tmp/work/_WORK.TEST.elab.so /tmp/work/_WORK.TEST.elab.o
** Note: writing FST waveform data to test.fst

*** Caught signal 11 (SIGSEGV) [address=0x5614a6ace000, ip=0x7f19a354f254] ***

[0x7f19a354f254] (/opt/nvc/share/nvc/ieee/_IEEE.STD_LOGIC_1164-body.so) IEEE.STD_LOGIC_1164.RESOLVED(37IEEE.STD_LOGIC_1164.STD_ULOGIC_VECTOR)U
```

No success, try again.

```Shell
$ nvc -a test.vhdl -e test -r -w  test
/usr/bin/ld -shared -o /tmp/work/_WORK.TEST.elab.so /tmp/work/_WORK.TEST.elab.o
** Note: writing FST waveform data to test.fst

*** Caught signal 11 (SIGSEGV) [address=0x55e8bc5bb000, ip=0x7fe7361ef254] ***

[0x7fe7361ef254] (/opt/nvc/share/nvc/ieee/_IEEE.STD_LOGIC_1164-body.so) IEEE.STD_LOGIC_1164.RESOLVED(37IEEE.STD_LOGIC_1164.STD_ULOGIC_VECTOR)U
```

Again.

```Shell
$ nvc -a test.vhdl -e test -r -w  test
/usr/bin/ld -shared -o /tmp/work/_WORK.TEST.elab.so /tmp/work/_WORK.TEST.elab.o
** Note: writing FST waveform data to test.fst
```

A bit better, but all signals are tristate:

```Shell
$ fst2vcd test.fst 
$date
	Wed Apr 15 15:14:53 2020

$end
$version
	nvc 1.5-devel
$end
$timescale
	1fs
$end
$scope vhdl_architecture test $end
$var logic 1 ! foo $end
$upscope $end
$enddefinitions $end
$dumpvars
#0
z!
```

I suspect the reason I don't always get a SEGV is just luck (good or bad, I'm not sure).

```Shell
$ nvc  --version
nvc 1.5-devel (Using LLVM 8.0.1)
Copyright (C) 2011-2018  Nick Gasson
This program comes with ABSOLUTELY NO WARRANTY. This is free software, and
you are welcome to redistribute it under certain conditions. See the GNU
General Public Licence for details.
```

I suspect this is because of some incompatibility with something in my
environment (LLVM?), but I really have no idea how to debug
this. Running from within GDB doesn't seem to trigger the SEGV. I've
tried LLVM7, but failed to get NVC to compile against it.

Build log in attached text file.

[nvc.txt](https://github.com/nickg/nvc/files/4481531/nvc.txt)

nickg, 2020-04-16:
Thanks for the detailed report. I can reproduce this LLVM 8.0.1 too, but not 7. Commit 9ae0b6c5 should fix the particular crash you're seeing but the test suite has other failures with 8 so not sure how much further you'll get. What error did you get compiling against 7?

joda4711, 2020-04-16:
Quick response!

With compiling against llvm7 I get the following:

```Shell
$ ../configure --prefix=/opt/nvc7 --with-llvm=/opt/llvm-7.1.0
[...output identical to llvm8 build removed...]
  CXXLD    bin/nvc
  NVC      lib/std/STD.STANDARD
  NVC      lib/ieee/IEEE.STD_LOGIC_1164
  NVC      lib/ieee/IEEE.NUMERIC_BIT
  NVC      lib/ieee/IEEE.NUMERIC_STD
  NVC      lib/ieee/IEEE.STD_LOGIC_1164-body
  NVC      lib/ieee/IEEE.NUMERIC_STD-body
  NVC      lib/ieee/IEEE.NUMERIC_BIT-body
  NVC      lib/ieee/IEEE.MATH_REAL
  NVC      lib/ieee/IEEE.MATH_REAL-body
  NVC      lib/ieee/IEEE.MATH_COMPLEX
  NVC      lib/ieee/IEEE.MATH_COMPLEX-body
  NVC      lib/nvc/NVC.ENV
  NVC      lib/std/STD.TEXTIO
nvc: /tmp/llvm-7.1.0.src/lib/IR/Constants.cpp:972: static llvm::Constant* llvm::ConstantArray::getImpl(llvm::ArrayType*, llvm::ArrayRef<llvm::Constant*>): Assertion `V[i]->getType() == Ty->getElementType() && "Wrong type in array element initializer"' failed.

*** Caught signal 6 (SIGABRT) ***

[0x7f48d38c3d21] (/lib64/libc-2.29.so) ../sysdeps/unix/sysv/linux/raise.c:51 raise
[0x7f48d38ad534] (/lib64/libc-2.29.so) /var/tmp/portage/sys-libs/glibc-2.29-r7/work/glibc-2.29/stdlib/abort.c:79 abort
[0x7f48d38ad40e] (/lib64/libc-2.29.so) /var/tmp/portage/sys-libs/glibc-2.29-r7/work/glibc-2.29/assert/assert.c:92 __assert_fail_base.cold
[0x7f48d38bb711] (/lib64/libc-2.29.so) /var/tmp/portage/sys-libs/glibc-2.29-r7/work/glibc-2.29/assert/assert.c:101 __assert_fail
[0x5608f902bf22] /tmp/llvm-7.1.0.src/lib/IR/Constants.cpp:972 _ZN4llvm13ConstantArray7getImplEPNS_9ArrayTypeENS_8ArrayRefIPNS_8ConstantEEE
[0x5608f902bdec] /tmp/llvm-7.1.0.src/lib/IR/Constants.cpp:961 _ZN4llvm13ConstantArray3getEPNS_9ArrayTypeENS_8ArrayRefIPNS_8ConstantEEE
[0x5608f9058a67] /tmp/llvm-7.1.0.src/lib/IR/Core.cpp:1212 LLVMConstArray
[0x5608f76ac3a8] ../src/cgen.c:1388 cgen_op_image_map.isra.0
-->          LLVMValueRef init = LLVMConstArray(array_type, lvalues, map.nelems);
             LLVMSetInitializer(values_glob, init);
[0x5608f76ad79c] ../src/cgen.c:3049 cgen_code
       case VCODE_OP_IMAGE_MAP:
-->       cgen_op_image_map(i, ctx);
          break;
[0x5608f76b09d8] ../src/cgen.c:3176 cgen_subprograms
       cgen_locals(&ctx);
-->    cgen_code(&ctx);
       cgen_free_context(&ctx);
[0x5608f76b1167] ../src/cgen.c:3599 cgen
       cgen_reset_function(t);
-->    cgen_subprograms(vcode);
    }
[0x5608f763b543] ../src/nvc.c:196 process_command
             fbuf_close(fbuf);
-->          cgen(units[i], vu);
          }
[0x5608f7639caf] ../src/nvc.c:1061 main
-->    return process_command(argc, argv);
    }
make[1]: *** [Makefile:2513: lib/std/STD.TEXTIO] Error 2
make: *** [Makefile:909: all] Error 2
```

nickg, 2020-04-18:
The above commit should fix the assertion failure with LLVM 7

joda4711, 2020-04-21:
It does for me as well. Thanks.

Does all this mean that LLVM7 is the recommended thing to build
against? I don't know much about the internals of LLVM, but I assume
that if LLVM8 is different enough to not work, then I assume that's
true for 9 and 10 as well.

Perhaps this should go in a README (perhaps it's there and I just
didn't read it).

nickg, 2020-04-23:
> 
> Perhaps this should go in a README (perhaps it's there and I just
> didn't read it).

It's not but I've just added it, thanks for suggesting. I'll leave this open until I fix the issues with 8 & 9.

closed 2020-04-24 by nickg

nickg, 2020-04-24:
Should work on 8 and 9 with the above commit

commit f48c651e (2020-04-24): Do not use byval argument passing for arrays. Fixes #392

Causes crashes with LLVM 8+