nickg/nvc #395: ** Fatal: value -1 outside bounds 0 to 2147483647 for parameter R

fmhess, 2020-06-04:
Adding an integer with value -1 to an unsigned with value zero causes a fatal error at runtime.  Test case below.
```
-- testcase for nvc complaining about decrementing an unsigned to -1

library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity nvc_unsigned_underflow_testbench is
end nvc_unsigned_underflow_testbench;
     
architecture behav of nvc_unsigned_underflow_testbench is
begin
	process
		variable un: unsigned(15 downto 0);
		variable inc: integer;
	begin
		un := to_unsigned(0, 16);
		un := un - 1; -- works fine
		assert un = X"ffff";
		
		un := to_unsigned(0, 16);
		inc := -1;
		un := un + inc; -- Fatal error on execution.
		assert un = X"ffff";
	end process;
end behav;

```

nickg, 2020-06-04:
I think this is the correct behaviour. In `numeric_std.vhdl` the `"+"` operator is defined as

```vhdl
function "+" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
```

There's an implicit conversion from `integer` to `natural` when you pass `inc` as the second argument and the range check fails when `inc` is -1. 

The `"-"` operator is defined similarly

```vhdl
function "-" (L: UNSIGNED;R: NATURAL) return UNSIGNED;
```

But in the case of `un - 1` which is equivalent to `"+"(un, 1)` the second argument is +1 which is ok.

GHDL gives a similar error:

```
/usr/bin/ghdl-mcode:error: bound check failure at ../test/regress/issue395.vhd:21
in process .issue395(behav).P0
  from: process work.issue395(behav).P0 at issue395.vhd:21
/usr/bin/ghdl-mcode:error: simulation failed
```

fmhess, 2020-06-04:
Hi, thanks for explaining.  I noticed ieee.std_logic_arith has a definition for   
```
function "+"(L: UNSIGNED; R: INTEGER) return UNSIGNED;
```
so out of curiosity I tried the following, but it segfaults when executed.

```
library IEEE;
use ieee.std_logic_1164.all;
--use ieee.numeric_std.all;
use ieee.std_logic_arith.all;

entity nvc_unsigned_underflow_testbench is
end nvc_unsigned_underflow_testbench;
     
architecture behav of nvc_unsigned_underflow_testbench is
begin
	process
		variable un: unsigned(15 downto 0);
		variable inc: integer;
	begin
		un := (others => '0');
		inc := -1;
		un := un + inc;
	end process;
end behav;
```

fmhess, 2020-06-04:
Also, I'm curious why an integer should automatically convert to a natural?  natural is a subtype of integer, so I understand that a natural would automatically convert to an integer, but types shouldn't auto convert to subtypes right?

closed 2020-06-06 by nickg

nickg, 2020-06-06:
> so out of curiosity I tried the following, but it segfaults when executed.
>
> ```
> library IEEE;
> use ieee.std_logic_1164.all;
> --use ieee.numeric_std.all;
> use ieee.std_logic_arith.all;
>
> entity [...]

Thanks, I've fixed this now.

> Also, I'm curious why an integer should automatically convert to a natural? natural is a subtype of integer, so I understand that a natural would automatically convert to an integer, but types shouldn't auto convert to subtypes right?

I think as far as the standard is concerned, natural and integer are both subtypes of the type integer, and are implicitly convertible with a runtime check the value is in the correct range. For function calls the key text is in section 9.3.4 or the LRM:

> Evaluation of a function call includes evaluation of the actual parameter expressions specified in the call and evaluation of the default expressions associated with formal parameters of the function that do not have actual parameters associated with them. **In both cases, the resulting **value** must belong to the subtype of the associated formal parameter.**

fmhess, 2020-06-06:
I've never read the VHDL language spec, but what is confusing me is natural vs integer is being handled inconsistently with unsigned vs std_logic_vector.  To demonstrate what I mean:

```
library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity nvc_argconv_test is
end nvc_argconv_test;
     
architecture behav of nvc_argconv_test is
begin
	process
		function myfuncA(x: unsigned) return unsigned is
		begin
			return x + 1;
		end function;

		function myfuncB(x: natural) return natural is
		begin
			return x + 1;
		end function;

		variable nat_var: integer;
		variable int_var: integer;
		variable stl_var: std_logic_vector(7 downto 0);
		variable unsigned_var: unsigned(7 downto 0);
	begin
			int_var := 1;
			-- works unexpectedly, even though function parameter type 
			-- is natural, a subtype of the integer value type
			nat_var := myfuncB(int_var);

			stl_var := (others => '0');
			-- fails as expected, because function parameter type 
			-- is unsigned, a subtype of the std_logic_vector value type 
			unsigned_var := myfuncA(stl_var);
		end process;
end behav;

```

commit 19572bce (2020-06-06): Functions that allocate from temp stack cannot be pure. Fixes #395