nickg/nvc #397: cast to integer rounds down

fmhess, 2020-06-05:
I've read that casting a real to integer is supposed to round to nearest in VHDL.  However, it seems to round down in nvc.  This also seems to break the log function, as log2(8) returns a value slightly below 3.  The ieee.math_real.log function does a cast to integer and I think that is why the rounding bug breaks the log function:

```
       J := INTEGER(REAL(N)*(G-1.0)); -- C code adds 0.5 for rounding
```

Test case below.

```
library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

entity nvc_rounding_bug is
end nvc_rounding_bug;
     
architecture behav of nvc_rounding_bug is
begin
	process
		variable i: integer;
		variable r: real;
	begin

        r := 0.9;
        i := integer(r);
        assert i = 1 report real'image(r) & " cast to integer gives " & integer'image(i);

        r := floor(log2(real(8)));
        i := integer(r);
        assert i = 3 report "floor(log2(real(8))) cast to integer gives " & integer'image(i);

        end process;
end behav;
```

nickg, 2020-06-06:
Thanks for reporting this. I fixed the rounding mode but it doesn't help with `LOG2`: it still returns `3.0 - DBL_EPSILON` :-/

fmhess, 2020-06-06:
Maybe the log2(8) bug is in math_real, because the same bug exists in Quartus

https://www.intel.com/content/www/us/en/programmable/support/support-resources/knowledge-base/solutions/rd01132011_181.html

and Modelsim

https://www.edaboard.com/threads/sizing-a-vhdl-input-port.328618/

nickg, 2022-07-30:
Sorry it's taken me so long to get back to this. I'm pretty sure the log2 bug is in the math_real package itself, but I don't have the mathematical background to debug it further. GHDL has the same behaviour and so does Quartus and Modelsim according to your links above. I checked Aldec and strangely it does get exactly 3.0. *But* if I paste the definitions of `log2`, etc. from math_real into the test file and re-run, it gives the 2.9999 value. I'm guessing this means Aldec is using some optimised intrinsics to replace the original IEEE math_real source code. 

I'm going to close this now because the original rounding issue is fixed and there's not much I can do for the log2 problem.

closed 2022-07-30 by nickg