nickg/nvc #399: segfault during elaboration

fmhess, 2020-06-08:
Hi,

I get a segfault when elaborating the following test program:
```vhdl
library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity nvc_elab_segfault is
end nvc_elab_segfault;
     
architecture behav of nvc_elab_segfault is
	function floor_log2(arg: unsigned) return natural is
		variable floor_log2: natural;
	begin
		assert arg /= 0;
		
		for i in arg'length - 1 downto 0 loop
			if to_X01(arg(i)) /= '0' then
				floor_log2 := i;
				exit;
			end if;
		end loop;
		
		return floor_log2;
	end function;

begin
	process
		variable x: positive;
		variable u: unsigned(31 downto 0);
	begin
		-- assigning to u and then passing u works fine
		u := to_unsigned(8, u'length);
		x := floor_log2(u);
		
		-- this following line causes segfault during elaboration
		x := floor_log2(to_unsigned(8, u'length));
		wait;
	end process;

end behav;
```

fmhess, 2020-06-11:
The problem seems triggered by the to_X01 call.  If I remove it, the problem goes away.

fmhess, 2020-06-11:
The bug seems to be a line 497 of eval.c in eval_get_var
```
context_t *new_context = eval_new_context(state);
```
there is no error handling for when eval_new_context returns NULL, due to eval_alloc exhaustion.

nickg, 2020-06-12:
> The bug seems to be a line 497 of eval.c in eval_get_var
> 
> ```
> context_t *new_context = eval_new_context(state);
> ```
> 
> there is no error handling for when eval_new_context returns NULL, due to eval_alloc exhaustion.

Thanks, I've pushed a fix for this. It was missing a NULL check as you say.

commit 9938a730 (2020-06-12): Segfault when constant folding uses too much memory. Fixes #399
closed 2020-06-12 by nickg