nickg/nvc #401: ** Fatal: expression cannot be folded to an integer constant

fmhess, 2020-06-13:
The fix for issue 399 prevents segfault, but the following testcase still produces a fatal error on elaboration.

```vhdl
library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity nvc_constant_fold_fatality is
end nvc_constant_fold_fatality;
     
architecture behav of nvc_constant_fold_fatality is
	function floor_log2(arg: unsigned) return natural is
		variable result: natural := 0;
	begin
		assert arg /= 0;
		
		for i in arg'length - 1 downto 0 loop
			if to_X01(arg(arg'low + i)) /= '0' then
				result := i;
				exit;
			end if;
		end loop;
		
		return result;
	end function;

	constant log2_memory_data_width_in_bytes: positive := floor_log2(to_unsigned(8, 32));
	signal prefetch_start_address: unsigned(31 downto log2_memory_data_width_in_bytes);

begin
	process
	begin
		wait;
	end process;

end behav;
```

fmhess, 2020-06-17:
Would you be willing to accept a patch which fixes this problem by turning state->heap into a linked list of allocations rather than a fixed 16K allocation?  Or do you have a more ambitious refactor in mind?

fmhess, 2020-06-17:
Here is the patch I have in mind:

```
diff --git a/src/eval.c b/src/eval.c
index 240ec730..0ce67335 100644
--- a/src/eval.c
+++ b/src/eval.c
@@ -29,7 +29,6 @@
 #include <float.h>
 
 #define MAX_DIMS   4
-#define EVAL_HEAP  (16 * 1024)
 #define ITER_LIMIT 1000
 
 typedef enum {
@@ -78,14 +77,20 @@ struct context {
    int        refcount;
 };
 
+struct eval_alloc_struct;
+typedef struct eval_alloc_struct eval_alloc_t;
+struct eval_alloc_struct {
+   eval_alloc_t *next;
+   char mem[];
+};
+
 typedef struct {
    context_t   *context;
    int          result;
    tree_t       fcall;
    eval_flags_t flags;
    bool         failed;
-   void        *heap;
-   size_t       halloc;
+   eval_alloc_t *allocations;
    loc_t        last_loc;
    int          iterations;
 } eval_state_t;
@@ -344,20 +349,10 @@ static void eval_assert_fail(int op, value_t *value, const char *value_str,
 
 static void *eval_alloc(size_t nbytes, eval_state_t *state)
 {
-   if (state->halloc + nbytes > EVAL_HEAP) {
-      EVAL_WARN(state->fcall, "evaluation heap exhaustion prevents "
-                "constant folding (%zu allocated, %zu requested)",
-                state->halloc, nbytes);
-      state->failed = true;
-      return NULL;
-   }
-
-   if (state->heap == NULL)
-      state->heap = xmalloc(EVAL_HEAP);
-
-   void *ptr = (char *)state->heap + state->halloc;
-   state->halloc += nbytes;
-   return ptr;
+   eval_alloc_t *new_alloc = xmalloc(sizeof(eval_alloc_t) + nbytes);
+   new_alloc->next = state->allocations;
+   state->allocations = new_alloc;
+   return new_alloc->mem;
 }
 
 static bool eval_new_var(value_t *value, vcode_type_t type, eval_state_t *state)
@@ -473,6 +468,19 @@ static void eval_free_context(context_t *context)
    }
 }
 
+void eval_cleanup_state(eval_state_t *state)
+{
+   eval_alloc_t *next, *current;
+   
+   eval_free_context(state->context);
+   for (current = state->allocations; current != NULL; current = next)
+   {
+      next = current->next;
+      free(current);
+   }
+   state->allocations = NULL;
+}
+
 static value_t *eval_get_reg(vcode_reg_t reg, eval_state_t *state)
 {
    return &(state->context->regs[reg]);
@@ -508,15 +516,13 @@ static value_t *eval_get_var(vcode_var_t var, eval_state_t *state)
             .fcall   = state->fcall,
             .failed  = false,
             .flags   = state->flags | EVAL_BOUNDS,
-            .heap    = state->heap,
-            .halloc  = state->halloc
+            .allocations = state->allocations
          };
 
          eval_vcode(&new_state);
          vcode_state_restore(&vcode_state);
 
-         state->heap = new_state.heap;
-         state->halloc = new_state.halloc;
+         state->allocations = new_state.allocations;
 
          if (new_state.failed) {
             state->failed = true;
@@ -939,15 +945,13 @@ static void eval_op_fcall(int op, eval_state_t *state)
       .fcall   = state->fcall,
       .failed  = false,
       .flags   = state->flags | EVAL_BOUNDS,
-      .heap    = state->heap,
-      .halloc  = state->halloc
+      .allocations = state->allocations
    };
 
    eval_vcode(&new);
    vcode_state_restore(&vcode_state);
 
-   state->heap = new.heap;
-   state->halloc = new.halloc;
+   state->allocations = new.allocations;
 
    if (new.failed)
       state->failed = true;
@@ -2250,8 +2254,7 @@ tree_t eval(tree_t expr, eval_flags_t flags)
    thunk = NULL;
 
    if (state.failed) {
-      eval_free_context(state.context);
-      free(state.heap);
+      eval_cleanup_state(&state);
       return expr;
    }
 
@@ -2267,8 +2270,7 @@ tree_t eval(tree_t expr, eval_flags_t flags)
    }
 
    tree_t tree = eval_value_to_tree(&result, type, tree_loc(expr));
-   eval_free_context(state.context);
-   free(state.heap);
+   eval_cleanup_state(&state);
    return tree;
 }
 
```

nickg, 2020-06-18:
> Would you be willing to accept a patch which fixes this problem by turning state->heap into a linked list of allocations rather than a fixed 16K allocation? Or do you have a more ambitious refactor in mind?

I'm happy to accept patches either as pull requests here or by email (the output of `git format-patch` is best). 

I think a problem with the above change is that the heap area is used to return non-scalar objects from functions (e.g. a function like `to_unsigned()` that returns an array). So the memory shouldn't be freed immediately when the function returns. This can be checked with valgrind.

The math functions cause large allocations because of the constant tables `math_real` uses. One way around this would be to increase the size of the static allocation to something large enough. 64k? The number is mostly arbitrary, it's just to ensure there's a large-enough contiguous space and that the memory usage of the compile-time evaluation doesn't grow without bounds.

Normally the temporary heap allocations should be cleaned up by the `VCODE_OP_HEAP_SAVE` and `VCODE_OP_HEAP_RESTORE` opcodes (it's managed like a stack) but I just noticed these aren't implemented in `eval.c` which is probably a mistake. But I'm not sure if that would help in this case as the large allocations come from the global constant tables in `math_real`.

fmhess, 2020-06-18:
> I think a problem with the above change is that the heap area is used to return non-scalar objects from functions (e.g. a function like `to_unsigned()` that returns an array). So the memory shouldn't be freed immediately when the function returns. This can be checked with valgrind.

The patch doesn't change any logic regarding where the memory is freed.  All the allocations in the list are freed at the same points in the code where the old heap was freed.  I didn't check with valgrind, but if there are deallocation problems they were in the existing code before the patch.

fmhess, 2020-06-19:
> Normally the temporary heap allocations should be cleaned up by the `VCODE_OP_HEAP_SAVE` and `VCODE_OP_HEAP_RESTORE` opcodes (it's managed like a stack) but I just noticed these aren't implemented [...]

This testcase does not use math_real.  I am now defining a floor_log2 function for unsigned.  The memory usage increases as the length of the unsigned increases.

fmhess, 2020-06-19:
I tried valgrind, it shows the same vcode registry leak plus cgen warnings with the patch as without (assuming you jack EVAL_HEAP high enough so it doesn't fail to elaborate).

nickg, 2020-06-19:
> 
> The patch doesn't change any logic regarding where the memory is freed. All the allocations in the list are freed at the same points in the code where the old heap was freed. I didn't check with valgrind, but if there are deallocation problems they were in the existing code before the patch.

Yes you're right, sorry. The patch is fine. Can you make a pull request for it?

nickg, 2020-06-20:
Thanks for fixing this!

closed 2020-06-20 by nickg