nickg/nvc #405: Support for recursive entities

tomgdixon, 2020-11-15:
VHDL supports declaring recursive entities, or instatiations of a component within itself.  nvc freezes when trying to elaborate a design with recursive instatiation.  It would be great to support this, as it enables more generic use of code (adder trees, sorting networks, etc..).

nickg, 2021-10-03:
I'm sorry it took me so long to reply to this. Recursive entities have been supported for a while and there's an existing regression test for that. But I just found a bug with the `**` operator when I tried some more complex examples. This would have caused the hang you described. If you're still interested I'd be grateful if you could re-test with the latest master branch.

If the hang still occurs please try to get a stack trace by running `nvc -e` under gdb and then hit ctrl-c and type `bt`.

closed 2022-02-05 by nickg

tomgdixon, 2022-02-05:
Thanks.  I forgot to say that it does work now.
- Tom