nickg/nvc #409: Timestamps in _index cause nvc build to be non-reproducible

daym, 2020-12-16:
We're trying to package nvc in the GNU Guix distribution.

Timestamps in _index files causes nvc build to be non-reproducible (see https://reproducible-builds.org/ for why reproducibility is important).

We've tracked it down to writing out the timestamp of the "design unit" files. If the source files are compiled, then the timestamp of the design unit file is the time stamp of the system at that time. Therefore, if someone else compiles the same thing, the timestamp is likely be different, making the entire build of nvc non-reproducible (and diffoscope will flag the `_index` files as the culprit).

It would be nice to make the build reproducible somehow.

nickg, 2020-12-17:
The timestamp is not actually stored in the file: it's read each time the design unit is loaded with `stat(2)` - see `lib_get_aux()` in `src/lib.c`.

The `_index` file is sensitive to to the _order_ files were compiled in. Are you building with `make -j`? Does it make a difference if you don't use `-j`? I think that could be fixed by storing the entries in the `_index` file in sorted order.

daym, 2020-12-17:
Oh! That's good news!

Indeed, it's reproducible when disabling parallel builds.

Thank you.

nickg, 2020-12-17:
Should be fixed by the above commit. I did a few rounds of `make clean-libs && make -j && sha1sum lib/ieee/_index` and it always gets the same hash.

commit e37c73dc (2020-12-17): Deterministic order for library index. Fixes #409

Also fixed a few other potential errors or non-deterministic behaviour
when building libraries in parallel.
commit 6ba3e654 (2021-04-09): Deterministic order for library index. Fixes #409

Also fixed a few other potential errors or non-deterministic behaviour
when building libraries in parallel.
closed 2020-12-17 by nickg