nickg/nvc #410: Clarify which versions of VHDL this supports

Newbytee, 2021-03-10:
Greetings!

I'm adding this to the [list of HDL simulators](https://en.wikipedia.org/wiki/List_of_HDL_simulators) on Wikipedia, and what's not clear from reading the README of this repository is what versions of VHDL this supports. I gather that it supports a subset of VHDL-2008 and that you can use a library that implements VHDL-2008 features in VHDL-1997 if you need library functions from VHDL-2008, which implies that this supports VHDL-1997 as well. As someone new to VHDL (Maybe it's obvious to people familiar with VHDL that VHDL means VHDL-1997, or what IEEE 1076-2002 means exactly?) I would argue that figuring out which version(s) of VHDL this supports shouldn't require deduction like this, and that the README should be updated to clarify which versions are supported.

Thanks in advance, and I hope this doesn't come off as needlessly critical

nickg, 2021-03-11:
VHDL-X is the informal way of referring to the standard IEEE 1076-X. So nvc supports (almost) all of VHDL-2002 and the earlier standards -2000 and -1993 (but not -1987). There is some VHDL-2008 support but it's not complete. I agree the README could be clearer here.

nickg, 2021-06-21:
I rewritten the README and it now contains a [Language Support](https://github.com/nickg/nvc#language-support) section (although I may move most of this to the man page).

closed 2021-06-21 by nickg