nickg/nvc #413: tests fail to build: undefined symbol: _mark_point, etc

yurivict, 2021-04-14:
```
===>   nvc-1.5.1 depends on package: check>0 - found
gmake[1]: Entering directory '/disk-samsung/freebsd-ports/cad/nvc/work/nvc-r1.5.1'
gmake[2]: Entering directory '/disk-samsung/freebsd-ports/cad/nvc/work/nvc-r1.5.1'
test/test_lower.c:91:25: warning: format specifies type 'int' but the argument has type 'int64_t' (aka 'long') [-Wformat]
                 i, bb, e->value, vcode_get_value(i));
                        ^~~~~~~~
/usr/local/include/check.h:494:71: note: expanded from macro 'fail'
#define fail(...) _ck_assert_failed(__FILE__, __LINE__, "Failed" , ## __VA_ARGS__, NULL)
                                                                      ^~~~~~~~~~~
test/test_lower.c:91:35: warning: format specifies type 'int' but the argument has type 'int64_t' (aka 'long') [-Wformat]
                 i, bb, e->value, vcode_get_value(i));
                                  ^~~~~~~~~~~~~~~~~~
/usr/local/include/check.h:494:71: note: expanded from macro 'fail'
#define fail(...) _ck_assert_failed(__FILE__, __LINE__, "Failed" , ## __VA_ARGS__, NULL)
                                                                      ^~~~~~~~~~~
test/test_lower.c:107:25: warning: format specifies type 'int' but the argument has type 'int64_t' (aka 'long') [-Wformat]
                 i, bb, e->value, vcode_get_cmp(i));
                        ^~~~~~~~
/usr/local/include/check.h:494:71: note: expanded from macro 'fail'
#define fail(...) _ck_assert_failed(__FILE__, __LINE__, "Failed" , ## __VA_ARGS__, NULL)
                                                                      ^~~~~~~~~~~
3 warnings generated.
ld: error: undefined symbol: _mark_point
>>> referenced by test_util.c
>>>               test/test_util.o:(expect_errors)
>>> referenced by test_util.c
>>>               test/test_util.o:(test_error_fn)
>>> referenced by test_util.c
>>>               test/test_util.o:(test_error_fn)
>>> referenced by test_util.c
>>>               test/test_util.o:(run_elab)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_ident.c
>>>               test/test_ident.o:(test_ident_new_fn)
>>> referenced 2312 more times

ld: error: undefined symbol: _ck_assert_failed
>>> referenced by test_util.c
>>>               test/test_util.o:(expect_errors)
>>> referenced by test_util.c
>>>               test/test_util.o:(test_error_fn)
>>> referenced by test_util.c
>>>               test/test_util.o:(run_elab)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_util.c
>>>               test/test_util.o:(_parse_and_check)
>>> referenced by test_ident.c
>>>               test/test_ident.o:(test_ident_new_fn)
>>> referenced by test_ident.c
>>>               test/test_ident.o:(test_equality_fn)
>>> referenced by test_ident.c
>>>               test/test_ident.o:(test_istr_fn)
>>> referenced by test_ident.c
>>>               test/test_ident.o:(test_rand_fn)
>>> referenced 350 more times
```

OS: FreeBSD 12.2
clang-10

nickg, 2021-04-14:
`_ck_assert_failed` and `_mark_point` both come from the expansion of the macro `fail_if` in `check.h`. 

What version of check are you using? I saw the latest in FreeBSD ports is 0.15.2 which I think I've tested with.  

Can you attach the `config.log` file and full linker command line (`make V=1`) that fails?

yurivict, 2021-04-14:
check-0.15.2

yurivict, 2021-04-14:
https://people.freebsd.org/~yuri/nvc-config.log

nickg, 2021-04-15:
I tried building in a FreeBSD 12 VM and I couldn't reproduce this - either with the 1.5.1 tag or with the master branch. Is it possible there is some mismatch between the check version which the `check.h` header comes from and the version of the library it's linking against?

```
robbins:~/nvc/build $ git describe
r1.5.1
robbins:~/nvc/build $ ldd bin/unit_test
bin/unit_test:
        libexecinfo.so.1 => /usr/lib/libexecinfo.so.1 (0x800359000)
        libcheck.so.0 => /usr/local/lib/libcheck.so.0 (0x80035f000)
        libz.so.6 => /lib/libz.so.6 (0x80036c000)
        libm.so.5 => /lib/libm.so.5 (0x800388000)
        libthr.so.3 => /lib/libthr.so.3 (0x8003ba000)
        libc.so.7 => /lib/libc.so.7 (0x8003e7000)
        libelf.so.2 => /lib/libelf.so.2 (0x8007dd000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x8007f8000)
        librt.so.1 => /usr/lib/librt.so.1 (0x800812000)
robbins:~/nvc/build $ objdump -T /usr/local/lib/libcheck.so.0 | grep _mark_point
0000000000005ab0 g    DF .text  000000000000000a  Base        _mark_point
robbins:~/nvc/build $ uname -a
FreeBSD robbins 12.2-RELEASE-p4 FreeBSD 12.2-RELEASE-p4 GENERIC  amd64
```

nickg, 2021-04-15:
BTW I just pushed 845970a37d9 which avoids having to add `-lexecinfo` to LDFLAGS.

yurivict, 2021-04-15:
I've reran the tests after the system upgrade and they ran fine. Looks like it was some intermittent problem.

Thank you for your help, and for fixing the ```-lexecinfo```.

Yuri

closed 2021-04-15 by yurivict