nickg/nvc #414: Another SIGBUS crash

jklockars, 2021-04-19:
Great that you solved the other SIGBUS crash so quickly! 👍 
This nvc of yours has turned out to work far better than I ever hoped for, besides for these small hiccups. It will be interesting to eventually try it with a much larger project than the memory test I am currently working on.

Here's another, very similar, SIGBUS crash for you to look into. Much the same code as before.
(I made sure to pull from GitHub and rebuild nvc.)

johan@Johans-Mac-mini Sources % time nvc -a bug4.vhd -e bug4 -r
/usr/bin/gcc -bundle -flat_namespace -undefined dynamic_lookup -o /Users/johan/Sources/work/_WORK.BUG4.elab.so /Users/johan/Sources/work/_WORK.BUG4.elab.o

*** Caught signal 10 (SIGBUS) [address=0x1027ec000, ip=0x0] ***

[0x100b64bc4] 3   nvc                                 0x0000000100b64bc4 bt_sighandler + 324
[0x1a2019c44] 4   libsystem_platform.dylib            0x00000001a2019c44 _sigtramp + 56
[0x1026cb8a0] 5   _WORK.BUG4.elab.so                  0x00000001026cb8a0 :bug4$WORK.BUG4(BEHAVIORAL).line_29.WR_DATA(VV) + 268
[0x1026cba28] 6   _WORK.BUG4.elab.so                  0x00000001026cba28 :bug4$WORK.BUG4(BEHAVIORAL).line_29.WR_DATA(25IEEE.NUMERIC_STD.UNSIGNED25IEEE.NUMERIC_STD.UNSIGNED) + 240
[0x1026cbb18] 7   _WORK.BUG4.elab.so                  0x00000001026cbb18 :bug4:line_29 + 188
[0x100bddc1c] 8   nvc                                 0x0000000100bddc1c rt_run + 220
[0x100bdc420] 9   nvc                                 0x0000000100bdc420 rt_run_sim + 1268
[0x100b61674] 10  nvc                                 0x0000000100b61674 process_command + 4608
[0x100b61390] 11  nvc                                 0x0000000100b61390 process_command + 3868
[0x100b61600] 12  nvc                                 0x0000000100b61600 process_command + 4492
[0x100b601e0] 13  nvc                                 0x0000000100b601e0 main + 1028
[0x1a1fedf34] 14  libdyld.dylib                       0x00000001a1fedf34 start + 4
nvc -a bug4.vhd -e bug4 -r  0.05s user 0.01s system 90% cpu 0.065 total

Another issue showed up while I created this small version. This only shows up without any clock wait, and with really large loop counts:

zsh: segmentation fault  nvc -a bug4.vhd -e bug4 -r
[bug4.txt](https://github.com/nickg/nvc/files/6338588/bug4.txt)

nickg, 2021-04-21:
Thanks for reporting this again. The SIGBUS should be fixed now but the segfault is still there (really just running out of stack space).

I'm curious about the comments at the top of bug.txt - have you been doing some profiling?

jklockars, 2021-04-21:
Ooops! Did I leave those comments in there...

Yes, I have done a bit of profiling to figure out if it would be worth optimising some things in my "RAM". Not terribly important, but it was interesting to see how various things affected the run-time.

For simulation, I needed something that could deal with random accesses to anywhere in a memory space covering at least 56 bits of address, and capable of providing as much RAM as possible. Thus the test writing 400 MByte - which as the comments state currently uses around 1 GByte actual RAM in nvc (certainly good enough for my purposes, but I have not tested this on any other simulator yet), while still having "good enough" performance.

nickg, 2021-04-25:
OK, thanks for the information. I haven't tested with memories that large before. The segfault should be fixed now too.

closed 2021-04-25 by nickg